revtel/react-native-nfc-manager #1: How do I get the data of NFC card?

YKSing, 2017-08-01:
Currently I just get this tag info:
```
Object {
  "id": Array [
    4,
    -32,
    121,
    10,
    116,
    76,
    -128,
  ],
  "maxSize": 137,
  "techTypes": Array [
    "android.nfc.tech.NfcA",
    "android.nfc.tech.MifareUltralight",
    "android.nfc.tech.Ndef",
  ],
  "type": "NFC Forum Type 2",
}
```

How do I get the stored data from NFC card?

whitedogg13, 2017-08-01:
Hi YK,

Look likes you have a RTD URI record, for now, please use following code to parse your specific tag data, which is a url:
```
function parseUri(record) {
    const RTD_URI_TYPE = [85];
    const RTD_URI_SCHEMES = [
        '',
        'http://www.',
        'https://www.',
        'http://',
        'https://',
        'tel:',
        'mailto:',
        // only list the top items from spec
    ];

    let arrayEqual = (a, b) => a && b && a.length === b.length && a.every((v, i) => v === b[i]);
    let bytesToStr = bytes => bytes.reduce((acc, byte) => acc + String.fromCharCode(byte), '');
    if (record.tnf === 1) {
        if (record.type && arrayEqual(RTD_URI_TYPE, record.type)) {
            let payload = record.payload,
                scheme = RTD_URI_SCHEMES[payload[0]];
            if (scheme !== undefined) {
                return {
                    url: scheme + bytesToStr(payload.slice(1))
                }
            }
        }
    }
    return null;
}
```

and call it like this:
```
let {url} = parseUri(tag.ndefMessage[0]);
console.log('parseUri: ' + url);
```

It should print out a string: 'http://www.ebsl.hk', for your specific tag.

BTW, We will try to implement more NDEF parsing functionality directly into this library. Any contributions are also welcome!

YKSing, 2017-08-01:
Thank you very much. I love this library :)

closed 2017-08-01 by YKSing