revtel/react-native-nfc-manager #5: Error: Redefinition of RCTMethodInfo after installing the plugin

adirzoari, 2017-10-17:
Hello,
I opened a new react-native project (latest version V0.49) and after installing the plugin react-nfc-manager,
I'm trying to run on ios with XCode 9 (simulators I test:IPHONE 6,7,7s IOS 11) it failed with error:

[image]

when I run on android it works fine.

whitedogg13, 2017-10-26:
Hi, @adirzoari thanks for interesting in this library!
I have already recognized the issue, and will quickly update new version once I verified the solution.

whitedogg13, 2017-10-27:
Hi, this should be fixed by PR #4 
Thanks!

closed 2017-10-27 by whitedogg13