revtel/react-native-nfc-manager #6: Attempt to get length of null array on NfcManager.start()

haroal, 2017-10-24:
Hi,

When I run `NfcManager.start()`, an error occurs saying "Attempt to get length of null array" and pointing to your `NfcManager.java` line 70. It indicates that invoke function does not accept null param. FYI, I tested on multiple Android devices, with and without a callback function as `NfcManager.start()` param.

So, I changed `NfcManager.java` line 70 `callback.invoke` param to a string and it works (`callback.invoke("null");` instead of `callback.invoke(null);`). I know that it is not a solution at all, so I just notice you about it because I don't have time to fix it and to send you a pull request.

Regards.

adirzoari, 2017-10-26:
the same issue, did you find solution?

haroal, 2017-10-26:
As I said in my previous message, I had to replace : 
`callback.invoke(null);`
by
`callback.invoke("null");`

here https://github.com/whitedogg13/react-native-nfc-manager/blob/master/android/src/main/java/community/revteltech/nfc/NfcManager.java#L70

This makes it work but it is a temporary solution because we should not edit node_modules packages.

adirzoari, 2017-10-26:
I found solution.
add in android->app->src->AndroidMainfest.xml premission to nfc

>     <uses-permission android:name="android.permission.NFC" />

haroal, 2017-10-26:
Yes, I already did this but this didn't solve the problem for me.

whitedogg13, 2017-10-26:
Hi both, sorry for the late response.
I have recognized the issue and pushed a fix. Really thanks for reporting the issue!

haroal, 2017-10-26:
Thank you very much ! It works great !

closed 2017-10-26 by haroal