revtel/react-native-nfc-manager #12: Native Module cannot be null in iOS simulator

milon87, 2017-11-23:
can we run in the simulator?
[image]

whitedogg13, 2017-11-25:
Hi, thanks for spotting this. I will check it out and get back soon.

whitedogg13, 2017-11-25:
Hi, I have tested against iOS simulator for both 9.3+iPhone6 and 11.0+iPhone8, and none of them throw such errors.

By looking into your error stack trace above, have you ever run `react-native link react-native-nfc-manager` to link the native module? It looks like that you're missing native module^^

milon87, 2017-11-25:
yes, I have linked. in android, it works perfectly.

whitedogg13, 2017-11-25:
Can you check if the ios project is properly linked?
1. the library project has been added into your `Libraries` group
2. the static lib `libNfcManager.a` should show in the `Build Phases -> Link Binary With Libraries` section.

If they are not appeared, you can manually link this library for now by following this instruction: https://facebook.github.io/react-native/docs/linking-libraries-ios.html

milon87, 2017-11-28:
the problem is here. anything I missing?

[image]

whitedogg13, 2017-11-29:
Hi, I still cannot reproduce this error. Can you check which XCode version you currently use? NFC related functionality only available for XCode 9.

milon87, 2017-11-29:
OS:  Sierra 10.12.6
XCode Version: 8.3.1 (8E1000a) 
ok in that case I will update to 9 and i will let you know

whitedogg13, 2017-12-09:
Hi @milon87 any update for this one? I think the XCode 9 is definitely required for NFC development. Hope your upgrading process goes smoothly~

closed 2017-12-14 by milon87

milon87, 2017-12-15:
This happens because of Xcode version. 
required version 9