revtel/react-native-nfc-manager #13: iPhone implementation crashing

rstims, 2017-12-03:
Hello. I've setup my code based on the provided example. Im running the app on my iPhone 7+ and followed the instructions on the video.  But when I trigger _startTrigger the app crashes and I cant seem to find an error.  Any help would be greatly appreciated.

whitedogg13, 2017-12-03:
Hi, thanks for trying this library.

The issues sounds like a native crash to me. Can you provide the native log using XCode? We can then further analyze the issue~

Thanks!

rstims, 2017-12-03:
@whitedogg13 thanks for your help, here is the console output..

```
2017-12-03 03:08:30.252017-0500 App[4913:3592503] NfcManager created
2017-12-03 03:08:30.310669-0500 App[4913:3592503] refreshPreferences: HangTracerEnabled: 0
2017-12-03 03:08:30.310808-0500 App[4913:3592503] refreshPreferences: HangTracerDuration: 500
2017-12-03 03:08:30.310824-0500 App[4913:3592503] refreshPreferences: ActivationLoggingEnabled: 0 ActivationLoggingTaskedOffByDA:0
2017-12-03 03:08:31.152 [info][tid:com.facebook.react.JavaScript] Running application "App" with appParams: {"rootTag":1,"initialProps":{}}. __DEV__ === true, development-level warning are ON, performance optimizations are OFF
2017-12-03 03:08:31.152493-0500 App[4913:3592589] Running application "App" with appParams: {"rootTag":1,"initialProps":{}}. __DEV__ === true, development-level warning are ON, performance optimizations are OFF
2017-12-03 03:08:31.235 [info][tid:com.facebook.react.WebSocketModuleQueue][RCTSRWebSocket.m:520] SocketRocket: In debug mode.  Allowing connection to any root cert
2017-12-03 03:08:31.235244-0500 App[4913:3592565] SocketRocket: In debug mode.  Allowing connection to any root cert
2017-12-03 03:08:31.314365-0500 App[4913:3592503] [MC] Lazy loading NSBundle MobileCoreServices.framework
2017-12-03 03:08:31.314977-0500 App[4913:3592503] [MC] Loaded MobileCoreServices.framework
2017-12-03 03:08:31.316944-0500 App[4913:3592503] [MC] System group container for systemgroup.com.apple.configurationprofiles path is /private/var/containers/Shared/SystemGroup/systemgroup.com.apple.configurationprofiles
2017-12-03 03:08:31.940982-0500 App[4913:3592568] NfcManager initialized
2017-12-03 03:08:32.006 [info][tid:com.facebook.react.JavaScript] 'start OK', undefined
2017-12-03 03:08:32.006436-0500 App[4913:3592589] 'start OK', undefined
2017-12-03 03:08:32.071 [info][tid:com.facebook.react.JavaScript] connected to remotedev-server
2017-12-03 03:08:32.070619-0500 App[4913:3592589] connected to remotedev-server
2017-12-03 03:08:32.297084-0500 App[4913:3592566] [] nw_connection_get_connected_socket 4 Connection has no connected handler
2017-12-03 03:08:32.297139-0500 App[4913:3592566] TCP Conn 0x1c0170b00 Failed : error 0:61 [61]
2017-12-03 03:08:35.436 [info][tid:com.facebook.react.JavaScript] 'registerTagEvent OK', undefined
2017-12-03 03:08:35.436282-0500 App[4913:3592589] 'registerTagEvent OK', undefined
2017-12-03 03:08:35.434491-0500 App[4913:3592566] [] nw_connection_get_connected_socket 8 Connection has no connected handler
2017-12-03 03:08:35.437406-0500 App[4913:3592566] TCP Conn 0x1c0170f80 Failed : error 0:61 [61]
```

whitedogg13, 2017-12-04:
Hi, sorry for the late response.

It seems normal from the native log above. Can you describe some more detail about the crash? Ex: is the crash happened is in JS or in native? any promise rejection? any yellow or red screen popup? or do you have JS log? 

Thanks!

rstims, 2017-12-05:
@whitedogg13 sorry for taking your time.  I ended up trying the example project in a brand new test project and it worked great.  Additionally, for anyone reading this, try it from a test project first, make sure it works there and then try to add it to your project.  Finally, I'll be using this module a lot.  Thank you.

closed 2017-12-05 by rstims

whitedogg13, 2017-12-06:
@rstims Thanks for clarifying & good to hear your project is working now!

arapocket, 2018-06-14:
Not having any value in the info.plist for the Privacy - NFC Scan Usage Description will cause it to crash.

But now that it's working, I found it does nothing on iOS! It reads the sticker apparently, but doesn't seem to see that there's any data in it.