revtel/react-native-nfc-manager #14: ios session closed in iphone 7

adirzoari, 2017-12-05:
Hello,
i'm trying to implement this plugin into my code and run nfc but I get these lines in console log
`registerTagEvent OK undefined
ios sessionClosed`

when I run the startDetection manually I get also the samei issues..
part of my code
`import { calcSize } from '../../utillities/utilites';
const { height, width } = Dimensions.get('window');

import NfcManager, { NdefParser } from 'react-native-nfc-manager';

export default class NfcScan extends Component {
    constructor(props) {
        super(props);
        this.state = {
            supported: true,
            enabled: false,
            tag: {},
            spinValue: new Animated.Value(0),
            opacityAnimate: new Animated.Value(1),
            width: new Animated.Value(calcSize(197)),
            nfcDetection: false
        }
    }

    componentDidMount() {
        // StatusBar.setHidden(true);

        NfcManager.start({
            onSessionClosedIOS: () => {
                console.log('ios session closed');
            }
        }).then(result => {
                if(!result){
                    console.log("not nfc detected");
                }else
                    console.log('start OK', result);
            })
            .catch(error => {
                console.warn('start fail', error);
                this.setState({supported: false});
            })

        if (Platform.OS === 'android') {
            NfcManager.getLaunchTagEvent()
                .then(tag => {
                    console.log('launch tag', tag);
                    if (tag) {
                        console.log(tag);
                        this.setState({ tag });
                    }
                })
                .catch(err => {
                    console.log(err);
                })
            NfcManager.isEnabled()
                .then(enabled => {
                    console.log(enabled);
                    this.setState({ enabled });
                })
                .catch(err => {
                    console.log(err);
                })
        }
    }`

whitedogg13, 2017-12-06:
Hi @adirzoari , thanks for adapting this library.

Regarding your questions:
1. Maybe you can try our sample to see whether it's working first. It's quite straightforward like this:
```javascript
// in your index.ios.js or index.android.js
import React, { Component } from 'react';
import {
  AppRegistry,
} from 'react-native';
import App from 'react-native-nfc-manager/example/App'

AppRegistry.registerComponent('NfcManagerDev', () => App);
```
2. If you still have problems integrating this library, you can post your full code (including the part calling `regsiterTagEvent`, which is missing in your code snippet above)

Thanks!

adirzoari, 2017-12-06:
hey, 
I looked into the example I don't see any mention of registerTagEvent or any use of it.
I test your example and it still same issue. 
I tested it with some real phone of iphone 6,7 and galaxy. all of them has nfc.

it's all my code  ( I also don't have any registerTagEvent and also in your example into github I don't see any in the registerTagEvent
`import React, { Component } from 'react';
import {
    View,
    Text,
    Image,
    ImageBackground,
    Platform,
    TouchableOpacity,
    Linking,
    StatusBar,
    Animated,
    Easing,
    StyleSheet,
    Dimensions
} from 'react-native';
// import Svg,{
//     Circle,
// } from 'react-native-svg';

import { calcSize } from '../../utillities/utilites';
const { height, width } = Dimensions.get('window');

import NfcManager, { NdefParser } from 'react-native-nfc-manager';

export default class NfcScan extends Component {
    constructor(props) {
        super(props);
        this.state = {
            supported: true,
            enabled: false,
            tag: {},
            spinValue: new Animated.Value(0),
            opacityAnimate: new Animated.Value(1),
            width: new Animated.Value(calcSize(197)),
            nfcDetection: false
        }
    }

    componentDidMount() {
        // StatusBar.setHidden(true);

        NfcManager.start({
            onSessionClosedIOS: () => {
                console.log('ios session closed');
            }
        }).then(result => {
                if(!result){
                    console.log("not nfc detected");
                }else
                    console.log('start OK', result);
            })
            .catch(error => {
                console.warn('start failed', error);
                this.setState({supported: false});
            })

        if (Platform.OS === 'android') {
            NfcManager.getLaunchTagEvent()
                .then(tag => {
                    console.log('launch tag', tag);
                    if (tag) {
                        console.log(tag);
                        this.setState({ tag });
                    }
[... 252 lines cut ...]
        fontSize: calcSize(83),
        color: '#ffffff',
        fontFamily: 'Montserrat-Regular',
        backgroundColor: 'transparent'
    },
    sloganText: {
        fontFamily: 'Montserrat-Regular',
        color: '#ffffff',
        fontSize: calcSize(28)
    },
    bottomContainer: {
        flex: 1,
        marginTop: calcSize(50),
        justifyContent: 'space-around',
        alignItems: 'center',

    },
    bottomText: {
        fontFamily: 'Montserrat-Regular',
        color: '#ffffff',
        fontSize: calcSize(28),
        fontWeight: '400'
    },

})`

adirzoari, 2017-12-07:
my question is if I need to implement and follow the youtube tutorial?
If i make react native app, do i need to write all the codes the guy from youtube write into swift file in xcode?

whitedogg13, 2017-12-08:
Hi, sorry for the late response.

You don't need to write any objective-c or swift code to use this library. However, you will need to do some project setup, please refer to `Prerequisites` section in following link:
* https://github.com/hansemannn/iOS11-NFC-Example

If the project is properly setup, you will see the NFC scanning you call `_startDetection`.

whitedogg13, 2018-01-12:
As mentioned in #23 , iOS seems working now, so I close this for now.

closed 2018-01-12 by whitedogg13