revtel/react-native-nfc-manager #16: NFC permission required even when set to not required in AndroidManifest.xml

troelsh, 2017-12-08:
Hi There,

After adding this module, we now have 14k unsupported devices because of the NFC requirement. I have tried adding the required false to the uses-permission in the android manifest, but it seems like it somehow gets overwritten?

We have wrapped the plugin so we only serve the function with the plugin if the device has NFC but still Play console deems it unsupported for devices without NFC.

<uses-permission android:name="android.permission.NFC" android:required="false" />

[image]

Any ideas?

troelsh, 2017-12-08:
Just found this little thing - trying to built with that in main mainfest file.

`<uses-feature android:name="android.hardware.nfc" android:required="false" />`

whitedogg13, 2017-12-08:
Hi @troelsh , really thanks for spotting this issue!
I will modify the `AndroidManifest.xml` based on this soon~

troelsh, 2017-12-09:
@whitedogg13 sounds great!

I wonder how i could test it before you build the next release other than manually editing the manifest in node_modules?

troelsh, 2017-12-09:
@whitedogg13 but this reversely might not work for app's that demands NFC functionality and only want NFC devices to see the app.

Maybe this setting should be removed from the modules maninfest and added as an instruction on how to set it up in the main manifest file of the app?

whitedogg13, 2017-12-09:
Hi, @troelsh thanks for the insight.

Regarding this topic, I think setting the requirement level to `false` is a fair option. The main reason for that is because NFC is used as a supplementary feature for most apps. 

Ultimately I think how to set this flag depends on the nature of the library. For example, I have checked another project I have contribute to (react-native-ble-manager), and in that project the hardware requiremen has been set to `true`. I think that's because BLE is mostly used as a mandatory feature. 

On the other hand, I think clarifying this point in documentation is definitely a good idea. Do you want to propose a PR for that? I'll be happy to merge it! :)

BTW, I have just added a small feature to detect NFC support by rejecting promise in `NfcManager.start` call in the latest version, this might affect your app if you're not handle promise rejection previously. Just for your reference.
 
Thanks!

troelsh, 2017-12-09:
@whitedogg13 i tried my best to make a suitable description that explains this. See PR #17 

I just found the bug i will let my dev colleague who has integrated your module know to handle promise rejections if he doesn't do that already. 👍

whitedogg13, 2017-12-13:
We now have set default hardware permission to false and PR #17  merged. Thanks @troelsh

closed 2017-12-13 by whitedogg13