revtel/react-native-nfc-manager #19: NfcManager.start return result 'undefined' always

devmoonsite, 2017-12-28:
Hello,
I installed the latest version you just published and I get result 'undefined' in any case I checked.
with undefined, I can't really check if the phone support in nfc or not, I don't have any indication to know..

whitedogg13, 2017-12-28:
Hi, thanks for trying this library!

The `start` call returns a promise, and the promise is **rejected** when the device doesn't support NFC. So in your case (promise **resolved** with undefined) means NFC is supported by your device.

devmoonsite, 2017-12-28:
thanks for quick response.
in iphone 4 it's returned me undefined and Iphone 4 not support nfc.
so maybe the Start function not works good?

whitedogg13, 2017-12-29:
Sounds like an issue, I will look into it and feedback later. Thanks for reporting!

adirzoari, 2017-12-29:
same issue. phone without support to nfc show me undefined and undefined as u explained above is resolve. not the good logic for me.

whitedogg13, 2018-01-02:
Hi both, just as you said, it's indeed a bug.
I have just fixed this issue in the latest version (v0.0.9), thanks for reporting!

closed 2018-01-02 by whitedogg13