revtel/react-native-nfc-manager #21: How to get NFC tag id on iOS

Symous, 2018-01-06:
Hi,awesome lib(碉堡了，老哥)!

I have successfully scan a NFC with this lib on iPhone, but I can not find out the tag id . Is there any way we can get the that ? thanks! :D

whitedogg13, 2018-01-07:
Hi, it's great to hear that you build an app using this lib! (很少在github上可以看到中文, 感覺很親切~)

Apple strongly limits the NFC usage to NDEF only, violating that will result in app rejection. BTW, previously there's a private API to get the ID, but has been disabled since iOS 11.2 so it seems not possible anymore. 

Similar discussions can be found here:
* https://github.com/chariotsolutions/phonegap-nfc/issues/286
* https://github.com/hansemannn/iOS11-NFC-Example/issues/16

Thanks!

Symous, 2018-01-08:
Thanks, my friend. :)

closed 2018-01-08 by Symous

changtimwu, 2018-04-14:
Can you guys show what tags do you use that can be read by iPhone?  I tried every tag I could grab near by.  None of them work. 😭

changtimwu, 2018-04-15:
感謝迅速回答.  NXP official iOS App 也是讀不到.
之前是隨便試, 悠遊卡, 門禁卡,  starbucks 隨行卡, Nordic Thingy-52,  後面兩個在 Android 讀起來資訊顯示是有 NDEF 的啊?  我訂了 NXP tag NTAG-215 明天試試看