revtel/react-native-nfc-manager #22: get feedback from user if he canceled nfc or not have permission for nfc

devmoonsite, 2018-01-10:
Hello,
I'm trying to get feedback from the nfc plugin if the user cancel or not give permission for nfc, can I do that?
I see you have support to know if the phone support nfc but I want to know if the phone support nfc and also have allow permission for use nfc.

whitedogg13, 2018-01-11:
Hi, thanks for feedback! 
Currently it is not implemented, but I think it is a good feature request and I will check it out when available.

devmoonsite, 2018-01-11:
okay, do you think you can do it soon?

whitedogg13, 2018-01-11:
Hi, after some survey I finally realize that we actually don't need to do runtime permission request for NFC feature, which means it is granted by default for both iOS and Android. 

You can simply verify this by downloading new apps with NFC feature, and see whether these apps fire any NFC related permission request.

I think what bothering you might be related to the permission request system introduced from Android 6.0. However, only features consider `dangerous` (such as contact or location) need to fire runtime permission request, others can still simply declare them in AndroidManifest.xml.

Hope that helps!