revtel/react-native-nfc-manager #23: using nfc with android

adirzoari, 2018-01-11:
Hello,
I'm trying to use in nfc with phone that support it. what I did it

`
componentDidMount(){  
NfcManager.start({
            onSessionClosedIOS: () => {
                console.log('ios session closed');
            }
        }).then(result => {
            console.log(result)
            this.setState({ nfcSupport: true })
        }).catch(err => {
            this.setState({ nfcSupport: false })

        })
        if (Platform.OS === 'android'){
            NfcManager.registerTagEvent(tag => {
                console.log('Tag Discovered', tag);
            }, 'Hold your device over the tag', true)
            NfcManager.isEnabled()
            .then(enabled => {
                console.log(enabled)
                this.setState({ nfcSupport: enabled ? true : false, nfcAllow: enabled ? true : false })

            })
            .catch(err => {
                console.log(err);
            })}`

but I don't see any response or any listener change when the nfc is work.
in ios it's works fine.

whitedogg13, 2018-01-12:
Hi, I have tested your code and it seems OK for me. (Unlike iOS, Android won't popup any system dialog during tag scanning, and we're considering to provide one to align the UX with iOS)

If the issue persists, maybe you can try to use `example/App.js` directly to see if it works?

adirzoari, 2018-01-12:
Yes i know it wont popup any dialog in android. When i navigate to the screen i run nfc, i hear the voice of the nfc but it do nothing. The registereventTag function should work in android too right?

whitedogg13, 2018-01-12:
Yes, it works in Android as well.

adirzoari, 2018-01-12:
when it scan nfc tag, what function should be run? in ios I see it run _startDetection function, but in android it not run it.

whitedogg13, 2018-01-12:
Hi, if you're using `example/App.js`, then calling `_startDetection` will start to listen to NFC tag, and after that `_onTagDiscovered` will be called when tags are found. Both iOS and Android have the same logic.

ingvardm, 2018-01-27:
Doesnt work for me either.
"react-native-nfc-manager": "^0.1.0",
"react": "16.0.0-beta.5",
"react-native": "0.49.3",

works on ios like a charm.

Any help is appreciated, thanks

whitedogg13, 2018-01-27:
Hi, can you describe more on this? I still cannot reproduce this issue currently. Maybe you can try to use the `example/App.js` in this repo and feedback the result.
Thanks!

andreap70, 2018-02-02:
Hi, i try to use your example but i obtain this error when i read a tag  (android device):
registerTagEvent fail [TypeError: undefined is not an object (evaluating 'NativeNfcManager.registerTagEvent')]
how can i solve this issue?
thank you

whitedogg13, 2018-02-02:
Hi, sounds like you cannot obtain `NativeNfcManager` in your app. Did you ever perform `react-native link react-native-nfc-manager` to link the native part into your app?

andreap70, 2018-02-03:
thank you for reply.
i use your example without change 
and i installed react-native-nfc-manager.

whitedogg13, 2018-02-04:
Hi, @andreap70 after `npm install`, you need to do an extra step: `react-native link react-native-nfc-manager`, otherwise the native part of this library will not be integrated into you project, which causes `NativeNfcManager` to be undefined.

andreap70, 2018-02-04:
i try to link react-native-nfc-manger but i obtain this:
_`react-native link` can not be used in Create React Native App projects. 
If you need to include a library that relies on custom native code, 
you might have to eject first. 
See https://github.com/react-community/create-react-native-app/blob/master/EJECTING.md 
for more information._
it's correnct? I don't know this command: 
reading docs, i see that i can eject project using "npm run eject", but i don't understand what happen before. 
I'm apologize, this is my first app with expo.io

whitedogg13, 2018-02-05:
Hi, @andreap70 no worry, it's totally fine to fire issues.

Regarding your question, yes, you will need to `npm run eject` first if you're using expo. Actually this step is required when installing ANY 3rd party library with native codes. 

Or, you can try to use the react native cli, check the `Building Projects with Native Code` tab in official link:
* https://facebook.github.io/react-native/docs/getting-started.html

whitedogg13, 2018-03-01:
Close this issue for now since no further discussions

closed 2018-03-01 by whitedogg13

Mjunaidkhan1, 2021-02-27:
Can anyone tell me how to use react-native-nfc-manager for Android app?
I am following documentary , and try v2-ios+android-read-ndef this example but my app not showing disply of ready for scanning after pressing test button, how can i do for android