revtel/react-native-nfc-manager #24: compileSdkVersion should bump to 26

YKSing, 2018-01-15:
According to the document from Google:
https://android-developers.googleblog.com/2017/12/improving-app-security-and-performance.html

And also CRNA project eject to Expokit are also default to 26:
https://github.com/expo/expo/commit/e4b0ff087ffbf3c3092165bf8397ecdb820d2b82

whitedogg13, 2018-01-16:
Hi, thanks for the suggestion! I will check if how this config affects the actual project and feedback later.

54vanya, 2018-07-11:
Hello, I can't compile release app without manually set compileSdkVersion 26, buildToolsVersion "26.0.2",  in library

JoeDurand3, 2018-07-27:
Will this library not work with version 26+? Right now none of my registerTagEvent functions are firing, hopefully this is not why. (I'm using Version 27)

whitedogg13, 2018-07-27:
Hi @JoeDurand3 @54vanya I plan to advance the compileSdkVersion and buildToolsVersion to 26 / 26.0.x in next release.

However, regarding your issues, I don't feel that they are related. I have just tested some combinations with current library:

1. SDK 26 + build tool 26.0.1/2 
2. SDK 27 + build tool 27.0.2

Both debug / release builds are tested, all of them run correctly.

If the issues persist, maybe you can update the build or runtime error messages so we can further investigate this.

JoeDurand3, 2018-07-27:
Thanks @whitedogg13 . I'll open a separate issue for my problem.

whitedogg13, 2018-08-13:
Hi all, I have advanced the SDK & build tool to 26 to latest release 0.5.4. Thanks you all!

closed 2018-08-27 by whitedogg13