revtel/react-native-nfc-manager #25: with React Navigation

milon87, 2018-01-15:
HomeScreen: read nfc tag
 _onTagDiscovered = tag => {
        try {
            let url = this._parseUri(tag);
            if (url) {
                let nodeID = url.split('/');
                this.props.navigation.navigate("ScreenB", {nodeID: nodeID[nodeID.length - 1]});
            }
        } catch (error) {
        }
    };

it reads fine and propagates to next screen until app goes in background and return to the foreground(android). Suppose I press back button and app goes in the background. when I reopen the app it reads well but can't propagate to the next screen. 

Warning I am getting: can only update a mounted or mounting components. This usually means you called setState, replaceState or forceUpdate .....

any issue with react navigation or nfc manager? 

Btw thanks for your effort. any suggestion is welcomed :)

milon87, 2018-01-16:
I have to add 
    componentWillUnmount() {
        NfcManager.unregisterTagEvent();
    }

otherwise it will not work

closed 2018-01-16 by milon87

whitedogg13, 2018-01-16:
Hi @milon87 thanks for reporting and also the proposed solution. 

Android's NFC discovery system is coupled with `foreground dispatch` logic, so it's indeed related to Activity resume/pause, just like you mentioned. So I think there might be a potential issue in current implementation and I will try to reproduce this one and feedback once I'm available.

BTW, also thanks for your contributions and I'm always glad to help!