revtel/react-native-nfc-manager #26: NfcManager.registerTagEvent got 1 arguments, expected 3

milon87, 2018-01-16:
from code
` _startDetection = () => {
        NfcManager.registerTagEvent(tag => {
            console.log('Tag Discovered', tag);
        }, 'Hold your device over the tag', true)
    };`

debug error:
NfcManager.registerTagEvent got 1 arguments, expected 3

`com.facebook.react.bridge.JavaMethodWrapper.invoke JavaMethodWrapper.java:344
com.facebook.react.bridge.JavaModuleWrapper.invoke JavaModuleWrapper.java:166
com.facebook.react.bridge.queue.NativeRunnable.run NativeRunnable.java
android.os.Handler.handleCallback Handler.java:739
android.os.Handler.dispatchMessage Handler.java:95
com.facebook.react.bridge.queue.MessageQueueThreadHandler.dispatchMessage 
      MessageQueueThreadHandler.java:31
android.os.Looper.loop Looper.java:155
com.facebook.react.bridge.queue.MessageQueueThreadImpl$3.run 
      MessageQueueThreadImpl.java:194
java.lang.Thread.run Thread.java:818`

SDK:27 
BuildTool: 27.0.2
React native: 0.46.4
"react-native-nfc-manager": "0.1.0",

whitedogg13, 2018-01-16:
Hi, thanks for reporting! It's indeed a bug, and I will fix it and feedback later.

whitedogg13, 2018-01-17:
Hi, I previously thought this might be related to the different native API signatures (iOS has extra options) for `registerTagEvent`, however they're actually identical (both taking three arguments, even through Android uses only one of them). So currently I am not sure why it happens and I also cannot reproduce it.

From the log you attached, it looks like JavaScript only pass 1 arg into native side, but I cannot find any code path might cause this (the only place to call native registerNfcTag is from JS registerNfcTag in line 83@NfcManager.js, but it passes 3 args). Are you help to check your `node_modules/react-native-nfc-manager/NfcManager.js` to see how the `registerTagEvent` is called? Maybe this file has be corrupted somehow?

Thanks

milon87, 2018-01-17:
the library is 100% ok 👍  . it was my mistake, I have built a release apk with version 0.0.6 after that I have upgraded to 0.1.0. 

the problem is related to Android studio. I have deleted app/src/assets folder then npm install and react-native link. now it is working fine.

closed 2018-01-17 by milon87