revtel/react-native-nfc-manager #27: Feature Request: Add Support for Writing Tags (Android only)

n0wak, 2018-01-23:
I would like to be able to write, in addition to read, some basic tags from an Android phone (my understanding is such actions are not currently permitted on iOS.) Is there any plan to implement this? I wish I could help but my Android/java is lacking.

whitedogg13, 2018-01-24:
Thanks for interested in this lib!

Yes, we do have plan to add tag-writing feature! I will keep this issue open and post any updates.

whitedogg13, 2018-01-29:
Hi, I have implemented a draft version for this feature, and now it can successfully write NDEF in my Android device. 

This implementation is currently in the `write-ndef` branch:
- https://github.com/whitedogg13/react-native-nfc-manager/tree/write-ndef

In this version, 2 more APIs `requestNdefWrite` and `cancelNdefWrite` are provided, and a full working example is also updated in `example/App.js`.

I will be appreciated if you can help to check it out and provide some feedback!

whitedogg13, 2018-02-03:
Hi, I just merge this ndef-writing feature and publish to npm (v0.2.0), a full example is also provided in the revised example along with this library.

I would like to hear any feedback from you, Thanks!

closed 2018-02-03 by whitedogg13