revtel/react-native-nfc-manager #30: Url is sent through requestNdefWrite() but isn't received by the listening device. The listening device receives a payload that isn't the url.

jstansbe, 2018-02-08:
I have successfully built/launched the app for android that was in the examples folder. I'm writing successfully to another device (which is also running the example) so i know the tag was sent. The problem is the payload is always http://play.google.com/store/apps/details?id=[sending app name]&feature=beam

The below console output is from the onTagDiscovered() method where i have also added some extra console.log lines to get more info. There are only 2 ndefmessages no matter what i do and neither contain what i was expecting (ie google.com). I thought there might be an error in the payload that was being sent out but when I tried using another nfc app (called NFC Tools) to send a URI that was reddit.com the result was the same. http://play.google.com/store/apps/details?id=[sending app name]&feature=beam.

Shouldn't the payload contain the info that i'm sending? 

[image]

whitedogg13, 2018-02-09:
Hi, thanks for adapting this library!

Regarding this issue, do you mean that you cannot see the exact ndef message you just write?

jstansbe, 2018-02-09:
My assumptions are that what is typed in the write textInput (google.com by default) is encoded in NDEF and sent out to listening devices. That is the info that isn't in the payload... unless im missing something?

whitedogg13, 2018-02-15:
Sorry to come back so late. However I cannot reproduce this issue, can you provide more information?

jstansbe, 2018-02-15:
Here is some of the code:
```
                    <View style={{width: '100%',padding: 10, marginTop: 20, borderWidth: 1, borderColor: 'black'}}>
                        <Text>NFC Data Exchange Format Test (Android)</Text>
                        <View style={{ flexDirection: 'row', alignItems: 'center' }}>
                            <Text>http://www.</Text>
                            <TextInput
                                style={{width: 200}}
                                value={urlToWrite}
                                onChangeText={urlToWrite => this.setState({ urlToWrite })}
                            />
                        </View>

                        <TouchableOpacity 
                            style={{ marginTop: 20, borderWidth: 1, borderColor: 'blue', padding: 10 }} 
                            onPress={isWriting ? this._cancelNdefWrite : this._requestNdefWrite}>
                            <Text style={{textAlign: 'center', color: 'blue'}}>{`${isWriting ? 'Cancel' : 'Write NDEF'} (Android Only) `}</Text>
                        </TouchableOpacity>
                    </View>
```

```
_requestNdefWrite = () => {
        function strToBytes(str) {
            let result = [];
            for (let i = 0; i < str.length; i++) {
                result.push(str.charCodeAt(i));
            }
            return result;
        }

        let {isWriting, urlToWrite} = this.state;
        if (isWriting) {
            return;
        }

        console.log("urlToWrite = " + urlToWrite);
        const urlBytes = strToBytes(urlToWrite);
        const headerBytes = [0xD1, 0x01, (urlBytes.length + 1), 0x55, 0x01];
        const bytes = [...headerBytes, ...urlBytes];
        console.log("bytes");console.log(bytes)
        this.setState({isWriting: true});

        NfcManager.requestNdefWrite(bytes)
            .then(() => console.log('write completed'))
            .catch(err => console.warn(err))
            .then(() => this.setState({isWriting: false}));
    }
```

whitedogg13, 2018-03-01:
Hi, sorry that I still cannot reproduce this issue, so I won't be able to fix it.

Just feel free to update any progress regarding this issue!

whitedogg13, 2018-03-13:
Close it for now due to no further activities

closed 2018-03-13 by whitedogg13