revtel/react-native-nfc-manager #31: Formatting NdefFormatable NFC tags

r0597192, 2018-02-14:
Hi,

I've been using your javascript library to make a basic NFC read/write app and it has worked well up untill now. My issue is that I need to format the tags to Ndef from NdefFormatable. Your library doesn't supply functions for this and the only alternative is porting a massive java library from chariotsolution's Phone-gap, which is not possible in my amount of given time (and skill).

Could you supply a method for formatting the tags to Ndef? Or give a reasoning I need to follow in order to code it myself.

Thanks

whitedogg13, 2018-02-15:
Hi, thanks for the feedback!

I'll love to implement this feature and will keep you posted.

whitedogg13, 2018-02-25:
Hi, sorry for the late response. I have just pushed first implementation regarding this feature onto the `format` branch, as well as the example:
* https://github.com/whitedogg13/react-native-nfc-manager/tree/format

However right now I don't have any tag with `NdefFormatable` tech, so I'm not able to test it. I will be very appreciated if you can help to check it out!

whitedogg13, 2018-03-13:
Hi, just tested this feature, and it works correctly in the latest 0.3.x release.
Thanks!

closed 2018-03-13 by whitedogg13

mahmoudhanoraa, 2020-11-07:
> Hi, sorry for the late response. I have just pushed first implementation regarding this feature onto the `format` branch, as well as the example:
>
> * [...]

This link is no longer working, can you please point out where the example is ?

ReactTalent, 2021-05-23:
> This link is no longer working, can you please point out where the example is ?

Same for me, are there any link to check this code? format NDEF.

kastwey, 2024-07-19:
Hi, @whitedogg13!
First of all, thanks so much for taking the time to write this awesome library! It helped me alot!
Regarding formating a tag to accept NDEF records, it seems to work only in android. Is it true? Could it be possible to do the same for iOS?

Thanks in advance!