revtel/react-native-nfc-manager #33: NfcManager is undefined (Android)

svvitale, 2018-02-16:
Hi @whitedogg13,

Thanks for all your work on this, I'm excited to start using it.  I've tried both your sample app and building some code into my own app and I'm seeing an issue where `NfcManager` is undefined:

```
import NfcManager, {NdefParser} from 'react-native-nfc-manager';

export default class Login extends React.Component {
  componentDidMount() {
    NfcManager.isSupported().then((isSupported) => Alert.alert(isSupported));
  }
}
```

When I run this app, I get this error:

```
TypeError: Cannot read property 'isSupported' of undefined
```

Any idea what I might be missing?

whitedogg13, 2018-02-18:
Hi, thanks for adapting this library!

Regarding your issue, do you remember to link the native part of this library? (ex, either by `react-native link react-native-nfc-manager` or by manually link)

svvitale, 2018-02-18:
Yes, I ran the link command and I see this in `settings.gradle`:

```
include ':react-native-nfc-manager'
project(':react-native-nfc-manager').projectDir = new File(rootProject.projectDir, '../node_modules/react-native-nfc-manager/android')
```

I also had to add it to my `build.gradle` file as a dependency to get it to build on Android:

```
dependencies {
    compile fileTree(dir: "libs", include: ["*.jar"])
    compile "com.android.support:appcompat-v7:23.0.1"
    compile "com.facebook.react:react-native:+"  // From node_modules
    compile project(':react-native-navigation')
    compile project(':react-native-nfc-manager')
}
```

whitedogg13, 2018-02-21:
Hi, sorry for the late response. We're still in Chinese New Year^^

Regarding this issue, I just test it again but I still cannot reproduce it. 

My flow is simple, I create a new RN project with `react-native init` and then install / link this library. After that, I use the example App.js, and it works correctly.

Can you help to check whether `NfcManager` or `NativeNfcManager` is undefined? 
* `NfcManager` is undefined: it should NEVER happen, since this lib is nothing more than a standard node_module. If this is the case, the issue might be:
    * please check whether your `node_modules/react-native-nfc-manager` is missing? Does it have correct structure (maybe you can simply compare to the repo) ?
    * please check if you have ever use `npm link` to cause npm to try to resolve `react-native-nfc-manager` to somewhere else? 
* `NativeNfcManager` is undefined: means the library is not properly linked, we can do more test to see where's wrong.

Thanks!

svvitale, 2018-02-24:
No problem at all, gong xi fa cai!

I've been able to further classify the issue.  If I begin a new react-native project and start with just `react-native-nfc-manager`, everything works perfectly.  It's when I pull in [react-native-navigation](https://github.com/wix/react-native-navigation) that the NFC manager stops working.  This is the install procedure I'm following: https://wix.github.io/react-native-navigation/#/installation-android

Do you see anything there that could be disrupting the NFC manager?

svvitale, 2018-02-24:
Aha!  Found it!  I didn't read this piece of `react-native-navigation` code carefully enough:

```
protected List<ReactPackage> getPackages() {
         // Add additional packages you require here
         // No need to add RnnPackage and MainReactPackage
         return Arrays.<ReactPackage>asList(
             // eg. new VectorIconsPackage()
         );
     }
```

I changed it to:
```
protected List<ReactPackage> getPackages() {
         // Add additional packages you require here
         // No need to add RnnPackage and MainReactPackage
         return Arrays.<ReactPackage>asList(
             new NfcManagerPackage()
         );
     }
```
And it all works perfectly.  Thanks for building this, and sorry for the false alarm!

closed 2018-02-24 by svvitale

whitedogg13, 2018-02-24:
@svvitale Cool! Good catch & thanks for sharing!

vivikenen, 2018-05-03:
@svvitale hi, 
Just wondering when you manually import, what package name should import on the top of the `MainApplication.java`?

What I mean is: 
```
protected List<ReactPackage> getPackages() {
         // Add additional packages you require here
         // No need to add RnnPackage and MainReactPackage
         return Arrays.<ReactPackage>asList(
             new NfcManagerPackage()
         );
     }
```
when you add the above code to `MainAppliacation.java`, do you need to import the related package on the top? (like `import com.facebook.react.ReactApplication;`

If you do, could I know the package name, please?  I could not found the package name.

whitedogg13, 2018-05-03:
@vivikenen It should be `community.revteltech.nfc`, let me know if you need other help!

whitedogg13, 2018-05-03:
@vivikenen fill import should be like this:

```java
import community.revteltech.nfc.NfcManagerPackage;
```

vivikenen, 2018-05-03:
Hi, thx for replying. It works now~ 
thx for your awesome package~
@whitedogg13