revtel/react-native-nfc-manager #35: NfcManager.requestNdefWrite(bytes) not returning a successful promise

Progoogler, 2018-03-06:
When I call `NfcManager.requestNdefWrite(bytes)` the promise never returns nor is an error ever caught. It just fires off the request and waits as though it's taking forever to write the Ndef. Is this the expected behavior? 

I imagine that it should complete immediately after requesting a write. Am I understanding this correctly, that `NfcManager.requestNdefWrite(bytes)` is creating the "radio wave" output of the `bytes` input, and once that process is completed successfully it should log "write completed"? If that is the case, can someone please point out the reason why this output is not generated?

```
  _startDetection = () => {
    NfcManager.registerTagEvent(this._onTagDiscovered)
    .then(result => {
        console.log('registerTagEvent OK', result)
        // Make sure I register tag event first
        if (Platform.OS === 'android') this._requestNdefWrite();
    })
    .catch(error => {
        console.warn('registerTagEvent fail', error)
    })
  }

  _requestNdefWrite = () => {
    function strToBytes(str) {
        let result = [];
        for (let i = 0; i < str.length; i++) {
          result.push(str.charCodeAt(i));
        }
      return result;
    }

    let { isWriting } = this.state;
    if (isWriting) {
        return;
    }

    const urlBytes = strToBytes('https://google.com');
    const headerBytes = [0xD1, 0x01, (urlBytes.length + 1), 0x55, 0x01];
    const bytes = [...headerBytes, ...urlBytes];

    this.setState({isWriting: true});
    NfcManager.requestNdefWrite(bytes)
      .then(() => console.log('write completed')) // Doesn't ever complete!
      .catch(err => console.warn(err)) // Logs "cancelled" after calling NfcManager.cancelNdefWrite()
      .then(() => {
        this.setState({isWriting: false})
        console.log('set state writing false') // Logs after calling NfcManager.cancelNdefWrite()
      });
  }

  _startDetection = () => {
    NfcManager.registerTagEvent(this._onTagDiscovered)
    .then(result => {
        console.log('registerTagEvent OK', result)
        if (Platform.OS === 'android') this._requestNdefWrite();
    })
    .catch(error => {
        console.warn('registerTagEvent fail', error)
    })
  }

  _stopDetection = () => {
    NfcManager.unregisterTagEvent()
    .then(result => {
        console.log('unregisterTagEvent OK', result)
        NfcManager.stop();
        if (Platform.OS === 'android') NfcManager.cancelNdefWrite();
    })
    .catch(error => {
        console.warn('unregisterTagEvent fail', error)
    })
  }

```

whitedogg13, 2018-03-06:
Hi, thanks for adapting this library.

Regarding your question, a successful NDEF write flow is like this:
1. Call `registerTagEvent` to start NFC tag detection.
2. Call `requestNdefWrite` to issue a write request, this will returns a pending promise, which means that the request is issued but not yet completed.
3. Now you can put NFC tag close to your Android device
4. When the tag is detected, the native part of this library will try to actually write NDEF into the tag
5. If the NDEF write process is completed, the pending promise (from step 2) will now be resolved. Otherwise it will be rejected with error.
6. After write request finished (no matter resolved or rejected), the library will resume to read tags and still fire NFC tag event when it reads any.

I think maybe you put NFC tag close to Android device too early, which means at that point there're no write request yet been issued.

Thank!

Progoogler, 2018-03-06:
Ah, I see! The promise resolves "on" writing the Ndef to the NFC tag. That explains it very well, thank you.

One extra question for clarification on the ability of this library. You mentioned that `requestNdefWrite` writes to a "tag" when it's detected. Does "tag" mean literally that this will only work with NFC tags or is it also a pseudonym for the NFC chip within another phone? The reason I ask is because I'm trying to transfer data between two mobile devices with this library. Is this a possibility?

whitedogg13, 2018-03-07:
Hi, thanks for the feedback!

As far as I know, the scenario you mentioned (two Android devices, one tries to write NDEF into another), might not work.

The reason for that is because normally the NFC-tag-mode system has two roles, the `tag` and the `reader`, and an Android device clearly acts as a reader rather than a tag. (There may be few Android devices can act as `virtual tag`, but there seems no public API available for us to implement)

So, in order to implement device communication via NFC, we should still utilize P2P mode. The P2P feature isn't currently supported by this library, and I hope to hear from you if this is a need (or, propose a PR, I will be glad to accept it^^)

Hope that helps!

Progoogler, 2018-03-07:
Wow, that clears things right up. Thank you!

P2P is definitely a feature that I think would bring a great benefit to developers and users alike. I read somewhere recently that NFC is nearing its plateau of trending rank since 2014 and that NDEF can transfer up to 2^32 bits of information. 

I would be glad to look into a PR. Will check back in!

closed 2018-03-07 by Progoogler