revtel/react-native-nfc-manager #38: undefined is not an object (evaluating 'NativeNfcManager.registerTagEvent')

SHBelsky, 2018-03-16:
I'm getting this error when trying to use this on a Google Pixel (1st generation regular, not XL). NFC is enabled on my phone; when I open the app or click on "Start Tag Detection", I get the below error.

```
[Unhandled promise rejection: TypeError: undefined is not an object (evaluating 'NativeNfcManager.isSupported')]
- node_modules/react-native-nfc-manager/NfcManager.js:95:6 in <unknown>
- node_modules/promise/setimmediate/core.js:45:7 in tryCallTwo
- node_modules/promise/setimmediate/core.js:200:23 in doResolve
- node_modules/promise/setimmediate/core.js:66:12 in Promise
- node_modules/react-native-nfc-manager/NfcManager.js:96:8 in isSupported
* null:null in componentDidMount
- node_modules/react-proxy/modules/createPrototypeProxy.js:61:45 in proxiedComponentDidMount
- node_modules/react-native/Libraries/Renderer/ReactNativeRenderer-dev.js:8979:12 in commitLifeCycles
- ... 22 more stack frames from framework internals
```

When I click "Start Tag Detection" I get a similar error:

```
10:50:34: undefined is not an object (evaluating 'NativeNfcManager.requestNdefWrite')
- node_modules/react-native-nfc-manager/NfcManager.js:41:3 in <unknown>
- node_modules/promise/setimmediate/core.js:45:7 in tryCallTwo
- node_modules/promise/setimmediate/core.js:200:23 in doResolve
- node_modules/promise/setimmediate/core.js:66:12 in Promise
- node_modules/react-native-nfc-manager/NfcManager.js:32:4 in requestNdefWrite
* App.js:136:29 in _requestNdefWrite
- node_modules/react-native/Libraries/Components/Touchable/TouchableOpacity.js:199:45 in touchableHandlePress
- node_modules/react-native/Libraries/Components/Touchable/Touchable.js:746:34 in _performSideEffectsForTransition
- ... 23 more stack frames from framework internals
```

Does that mean this package and my device are incompatible? I am attaching my `package.json` file to see if there are any problems with my setup that might stop this from working.

```
{
  "name": "nfctest",
  "version": "0.1.0",
  "private": true,
  "devDependencies": {
    "jest-expo": "25.0.0",
    "react-native-scripts": "1.11.1",
    "react-test-renderer": "16.2.0"
  },
  "main": "./node_modules/react-native-scripts/build/bin/crna-entry.js",
  "scripts": {
    "start": "react-native-scripts start",
    "eject": "react-native-scripts eject",
    "android": "react-native-scripts android",
    "ios": "react-native-scripts ios",
    "test": "node node_modules/jest/bin/jest.js"
  },
  "jest": {
    "preset": "jest-expo"
  },
  "dependencies": {
    "expo": "^25.0.0",
    "react": "16.2.0",
    "react-native": "0.52.0",
    "react-native-nfc-manager": "^0.3.1"
  }
}
```

whitedogg13, 2018-03-16:
Hi, thanks for adapting this library!

Looks like the native codes are not properly linked. For Expo project, you have to first `npm run eject` and then `react-native link react-native-nfc-manager`. Otherwise the native code won't work,

Thanks!

whitedogg13, 2018-03-28:
Hi, is there any update or anything I can help?

SHBelsky, 2018-03-28:
Sorry, meant to reply. Doing that fixed the issue. I wasn't aware that an eject was necessary. But now that I ejected, all is good. Thank you for checking in and helping me!

closed 2018-03-28 by SHBelsky

whitedogg13, 2018-03-28:
You're welcome!

jcpaesf, 2020-10-12:
> Hi, thanks for adapting this library!
> 
> Looks like the native codes are not properly linked. For Expo project, you have to first `npm run eject` and then `react-native link react-native-nfc-manager`. Otherwise the native code won't work,
> 
> Thanks!

Hi whitedogg13.

I used the react-native version 0.60.5 and, I do the expo eject, but I got the same issue. Can you help me?