revtel/react-native-nfc-manager #42: Can't write NFC TAG - message "you should requestTagEvent first"

pedrotainha, 2018-03-29:
Hi, i can read the tag properly, but i can't write.
It appear a warning "you should requestTagEvent first"

I'm using your APP as example to test before i make my app implementation.

I have my Huawei P8 connected as emulator.

If you need more information, let me know.

Thank you

whitedogg13, 2018-03-29:
Hi, thanks for trying this library!

As the warning says, you should call `requestTagEvent` API first before invoking NDEF write ( `requestNdefWrite` API). That's because we can only write to a tag after we detect it.

If you're using the example app, you can see a button named "Start Tag Detection", first click it then try to write NDEF again and it should work.

Thanks!

pedrotainha, 2018-03-29:
I thanks.

closed 2018-03-29 by pedrotainha