revtel/react-native-nfc-manager #43: Android + iOS NFC permissions

jerninvien, 2018-04-03:
Hi we are using this package and are very impressed.

Just was curious, is the package asking for permissions to use NFC on the device (Android / iOS) or should / can we add that manually?

e.g. on Android (AndroidManifest.xml):
```
<uses-permission android:name="android.permission.NFC" />
```

whitedogg13, 2018-04-03:
Hi, thanks for adapting this library. It's also great to hear that this lib is helpful!

Regarding the question:

* iOS: you will need to add NFC permission (NFCReaderUsageDescription) into your Info.plist (as well as some additional setup, this repo explains these setup pretty well: https://github.com/hansemannn/iOS11-NFC-Example), then iOS will prompt user when required. In JS side, you don't need any further setup.
* Android: you only need to add `<uses-permission android:name="android.permission.NFC" />` into AndroidManifest.xml, no need to use RN's `PermissionsAndroid` API, even after Android 6.0. That's because NFC isn't considered as `dangerous` permission and only dangerous ones need user approval (you can refer to this link to see the full list of dangerous permissions: https://developer.android.com/guide/topics/permissions/overview.html#permission-groups)

Hope that helps!

jerninvien, 2018-04-03:
Thanks very much, excellent answer!

whitedogg13, 2018-04-06:
It's good to help! I will close this issue for now.

closed 2018-04-06 by whitedogg13