revtel/react-native-nfc-manager #45: App crash with Null pointer Exception when press start  Tag detection in example

lapsonwor, 2018-04-12:
Hi, 
I have followed the Readme and trying the example.
The example screen showing however the app crashed.

My environment:
`environment:
  OS:  Android 4.2.1

Packages: (wanted => installed)
  react: 16.3.0-alpha.1
  react-native: 0.54.3
  react-native-nfc-manager: ^0.3.2`

` E/AndroidRuntime: FATAL EXCEPTION: mqt_native_modules
                                                                java.lang.NullPointerException
                                                                    at java.lang.AbstractStringBuilder.<init>(AbstractStringBuilder.java:86)
                                                                    at java.lang.StringBuilder.<init>(StringBuilder.java:95)
                                                                    at com.facebook.react.devsupport.DevSupportManagerImpl$JSExceptionLogger.log(DevSupportManagerImpl.java:303)
                                                                    at com.facebook.react.devsupport.DevSupportManagerImpl.handleException(DevSupportManagerImpl.java:287)
                                                                    at com.facebook.react.bridge.CatalystInstanceImpl.onNativeException(CatalystInstanceImpl.java:501)
                                                                    at com.facebook.react.bridge.CatalystInstanceImpl.access$900(CatalystInstanceImpl.java:40)
                                                                    at com.facebook.react.bridge.CatalystInstanceImpl$NativeExceptionHandler.handleException(CatalystInstanceImpl.java:517)
                                                                    at com.facebook.react.bridge.queue.MessageQueueThreadHandler.dispatchMessage(MessageQueueThreadHandler.java:33)
                                                                    at android.os.Looper.loop(Looper.java:137)
                                                                    at com.facebook.react.bridge.queue.MessageQueueThreadImpl$3.run(MessageQueueThreadImpl.java:194)
                                                                    at java.lang.Thread.run(Thread.java:856)`

Would you please let me know how to fix the issue. Thank you.

edy, 2018-06-14:
I'm getting the same error, but under other circumstances: I'm using HeadlessJS. If you call somewhere `await NfcManager.isSupported()` in your background task, the app crashes with the error message, posted by @lapsonwor. But only if the app is running in background.

whitedogg13, 2018-06-17:
@edy Thanks for reporting that. I will try to verify it and feedback later.

MickaToast, 2018-06-24:
I got the same error, I'm still using 0.0.7.
Here is the log:
`D/NfcManager(14730): start
D/NfcManager(14730): isEnabled
D/NfcManager(14730): registerTag
I/NfcManager(14730): enableForegroundDispatch, enable = true
W/dalvikvm(14730): threadid=14: thread exiting with uncaught exception (group=0x41677438)
E/AndroidRuntime(14730): FATAL EXCEPTION: mqt_native_modules
E/AndroidRuntime(14730): java.lang.NullPointerException
E/AndroidRuntime(14730):        at android.nfc.NfcAdapter.enableForegroundDispatch(NfcAdapter.java:1183)
E/AndroidRuntime(14730):        at community.revteltech.nfc.NfcManager.enableDisableForegroundDispatch(NfcManager.java:171)
E/AndroidRuntime(14730):        at community.revteltech.nfc.NfcManager.registerTagEvent(NfcManager.java:126)
E/AndroidRuntime(14730):        at java.lang.reflect.Method.invokeNative(Native Method)
E/AndroidRuntime(14730):        at java.lang.reflect.Method.invoke(Method.java:511)
E/AndroidRuntime(14730):        at com.facebook.react.bridge.JavaMethodWrapper.invoke(JavaMethodWrapper.java:374)
E/AndroidRuntime(14730):        at com.facebook.react.bridge.JavaModuleWrapper.invoke(JavaModuleWrapper.java:162)
E/AndroidRuntime(14730):        at com.facebook.react.bridge.queue.NativeRunnable.run(Native Method)
E/AndroidRuntime(14730):        at android.os.Handler.handleCallback(Handler.java:615)
E/AndroidRuntime(14730):        at android.os.Handler.dispatchMessage(Handler.java:92)
E/AndroidRuntime(14730):        at com.facebook.react.bridge.queue.MessageQueueThreadHandler.dispatchMessage(MessageQueueThreadH
andler.java:31)
E/AndroidRuntime(14730):        at android.os.Looper.loop(Looper.java:137)
E/AndroidRuntime(14730):        at com.facebook.react.bridge.queue.MessageQueueThreadImpl$3.run(MessageQueueThreadImpl.java:194)`

jperezsoy, 2018-07-16:
I got the same error.

`environment:
OS:  Android 4.1.2
Packages: (installed)
  react: 16.4.1
  react-native: 0.55.4
  react-native-nfc-manager: ^0.5.1
`
Here is the log:
`
07-16 16:30:17.975 16935-16960/com.testapp E/AndroidRuntime: FATAL EXCEPTION: mqt_native_modules
    java.lang.NullPointerException
        at java.lang.AbstractStringBuilder.<init>(AbstractStringBuilder.java:86)
        at java.lang.StringBuilder.<init>(StringBuilder.java:95)
        at com.facebook.react.devsupport.DevSupportManagerImpl$JSExceptionLogger.log(DevSupportManagerImpl.java:301)
        at com.facebook.react.devsupport.DevSupportManagerImpl.handleException(DevSupportManagerImpl.java:285)
        at com.facebook.react.bridge.CatalystInstanceImpl.onNativeException(CatalystInstanceImpl.java:511)
        at com.facebook.react.bridge.CatalystInstanceImpl.access$900(CatalystInstanceImpl.java:39)
        at com.facebook.react.bridge.CatalystInstanceImpl$NativeExceptionHandler.handleException(CatalystInstanceImpl.java:527)
        at com.facebook.react.bridge.queue.MessageQueueThreadHandler.dispatchMessage(MessageQueueThreadHandler.java:31)
        at android.os.Looper.loop(Looper.java:137)
        at com.facebook.react.bridge.queue.MessageQueueThreadImpl$3.run(MessageQueueThreadImpl.java:192)
        at java.lang.Thread.run(Thread.java:856)
07-16 16:30:18.785 16935-16935/com.testapp E/ActivityThread: Activity com.testapp.MainActivity has leaked IntentReceiver community.revteltech.nfc.NfcManager$1@41f9caa8 that was originally registered here. Are you missing a call to unregisterReceiver()?
    android.app.IntentReceiverLeaked: Activity com.testapp.MainActivity has leaked IntentReceiver community.revteltech.nfc.NfcManager$1@41f9caa8 that was originally registered here. Are you missing a call to unregisterReceiver()?
        at android.app.LoadedApk$ReceiverDispatcher.<init>(LoadedApk.java:798)
        at android.app.LoadedApk.getReceiverDispatcher(LoadedApk.java:593)
        at android.app.ContextImpl.registerReceiverInternal(ContextImpl.java:1325)
        at android.app.ContextImpl.registerReceiver(ContextImpl.java:1306)
        at android.app.ContextImpl.registerReceiver(ContextImpl.java:1299)
        at android.content.ContextWrapper.registerReceiver(ContextWrapper.java:430)
        at community.revteltech.nfc.NfcManager.start(NfcManager.java:274)
        at java.lang.reflect.Method.invokeNative(Native Method)
        at java.lang.reflect.Method.invoke(Method.java:511)
        at com.facebook.react.bridge.JavaMethodWrapper.invoke(JavaMethodWrapper.java:372)
        at com.facebook.react.bridge.JavaModuleWrapper.invoke(JavaModuleWrapper.java:160)
        at com.facebook.react.bridge.queue.NativeRunnable.run(Native Method)
        at android.os.Handler.handleCallback(Handler.java:615)
        at android.os.Handler.dispatchMessage(Handler.java:92)
        at com.facebook.react.bridge.queue.MessageQueueThreadHandler.dispatchMessage(MessageQueueThreadHandler.java:29)
        at android.os.Looper.loop(Looper.java:137)
        at com.facebook.react.bridge.queue.MessageQueueThreadImpl$3.run(MessageQueueThreadImpl.java:192)
        at java.lang.Thread.run(Thread.java:856)
`

christianchown, 2018-07-18:
Hello, I got the following related crash reported in Crashlytics:

```
Attempt to invoke virtual method 'android.content.pm.PackageManager android.app.Activity.getPackageManager()' on a null object reference

  | community.revteltech.nfc.NfcManager.isSupported (NfcManager.java:258)
  | java.lang.reflect.Method.invoke (Method.java)
  | com.facebook.react.bridge.BaseJavaModule$JavaMethod.invoke (BaseJavaModule.java:368)
  | com.facebook.react.cxxbridge.JavaModuleWrapper.invoke (JavaModuleWrapper.java:138)
  | com.facebook.react.bridge.queue.NativeRunnable.run (NativeRunnable.java)
  | android.os.Handler.handleCallback (Handler.java:739)
  | android.os.Handler.dispatchMessage (Handler.java:95)
  | com.facebook.react.bridge.queue.MessageQueueThreadHandler.dispatchMessage (MessageQueueThreadHandler.java:31)
  | android.os.Looper.loop (Looper.java:158)
  | com.facebook.react.bridge.queue.MessageQueueThreadImpl$3.run (MessageQueueThreadImpl.java:208)
  | java.lang.Thread.run (Thread.java:818)
```

I'm guessing that it came about as the app was backgrounded when we called `const NFCSupported = await NFCManager.isSupported();`, as in https://github.com/whitedogg13/react-native-nfc-manager/issues/45#issuecomment-397417651

Having looked at other Native Modules, it would seem that a null `getCurrentActivity()` needs guarding against.

whitedogg13, 2018-07-18:
@christianchown Thanks for the comment! It sounds reasonable and I will see what I can do.

BTW, I wonder how do you reproduce this error? Are you using headless JS as well?

christianchown, 2018-07-18:
Hi @whitedogg13 - we're not using headless JS, our crash log came from a vanilla React Native app. Our app does do some things - NFC tag handling, for one - when backgrounded though

whitedogg13, 2018-07-21:
Hi @christianchown I have added guardian code for this in the latest release (0.5.2), (simply test against getCurrentActivity() to avoid native crash) if you can give it a try I will be very appreciated!

christianchown, 2018-07-25:
Hi @whitedogg13, thanks for your changes. It all looks good, except in `enableDisableForegroundDispatch`, where there's:
https://github.com/whitedogg13/react-native-nfc-manager/blob/edfdae405d7f53f5949c82341db8406087e9ddab/android/src/main/java/community/revteltech/nfc/NfcManager.java#L445-L447

I got a Crashlytics crash log entry in this method. Can this be changed to:

```
Activity currentActivity = getCurrentActivity();

if (nfcAdapter != null && currentActivity != null && !currentActivity.isFinishing()) {
```

whitedogg13, 2018-07-25:
@christianchown Thanks in advance! I will update it soon.

christianchown, 2018-07-25:
@whitedogg13 thanks. Can these lines also be changed?
https://github.com/whitedogg13/react-native-nfc-manager/blob/edfdae405d7f53f5949c82341db8406087e9ddab/android/src/main/java/community/revteltech/nfc/NfcManager.java#L454-L456
I have been testing with an old device (a Galaxy Ace 3, Android KitKat) and it throws a `NullPointerException` in `enableForegroundDispatch` & `disableForegroundDispatch` when receiving a tag. The passed parameters seem fine, and operation works fine for our use after catching the Exception alongside `IllegalStateException`:
```
} catch (IllegalStateException | NullPointerException e) {
    Log.w(LOG_TAG, "Exception enabling or disabling NFC foreground dispatch.");
}
```

MickaToast, 2018-07-26:
@christianchown +1, I think it's related to my previous comment: https://github.com/whitedogg13/react-native-nfc-manager/issues/45#issuecomment-399759177

whitedogg13, 2018-07-27:
Hi @christianchown @MickaToast thanks for your feedback! I have just published a new version 0.5.3 addressing this bug according to prior discussion, please check it out to see whether the issue resolved.

whitedogg13, 2018-08-10:
Thanks! @vladimir-ivanov-epam

closed 2018-08-10 by whitedogg13