revtel/react-native-nfc-manager #51: Unable To write Plain text

sourav-dev-meme, 2018-04-16:
Need Help to write plain text. Is there any option where we can right plain text. Right now whatever we write is saved as URL.

Thanks in Advance !

whitedogg13, 2018-04-16:
Hi, I will check it out soon when I'm available :)

sourav-dev-meme, 2018-04-16:
Thanks for reverting back. 
Waiting for the solution :)

whitedogg13, 2018-04-20:
Hi, I have just updated the `example/App.js` to include a RTD_TEXT example. Just like the url example, it allows you to write arbitrary text as RTD_TEXT record in the latest commit, please check it out!

Source is [here](https://github.com/whitedogg13/react-native-nfc-manager/blob/master/example/App.js), you can refer to the `_requestNdefWrite` function.

Thanks!

whitedogg13, 2018-04-27:
Hi, @sourav-dev-meme how's going? Are there still any issues?

whitedogg13, 2018-06-17:
Close due to no further activity

closed 2018-06-17 by whitedogg13