revtel/react-native-nfc-manager #55: Is it possible to read the payload rather than just the tags itself

2joocy, 2018-04-26:
I am currently trying to work with React Native, where I have to process given data from a NFC chip. This chip has 2 types of data. One is the tags, the other is the datablocks itself in hex. Is it possible to access this data?

whitedogg13, 2018-04-27:
Hi, sorry for the late response.

I'm not sure that I understand your question. Do you mean that your data is not in NDEF format? Or you need a way to interpret your NDEF record payload?

Thanks

2joocy, 2018-04-27:
It supposedly is in the tag aswell as hex but is not showing

2joocy, 2018-04-27:
When using one of those scanning apps, you can ask from which data blocks you wanna read, or from which byte.

2joocy, 2018-04-27:
Another example is this blog post:

http://android-er.blogspot.dk/2015/10/android-nfc-readblock-for-mifareclassic.html

He reads the specific blocks, and displays them as hex data. Would this be possible here aswell?

poison, 2018-09-26:
Hi @2joocy,

I created PR #86, check it out and let me know what you think. It works for my usecase

Best,
Nicolas

poison, 2018-10-05:
Hi @2joocy,

If you check out PR #92 you can see some working sample code (slightly changed the interface as @whitedogg13 suggested on my previous PR)

Best,
Nicolas

whitedogg13, 2018-10-06:
Hi @2joocy , we have merged the PR by @poison and just published 0.7.0 release to support basic MifareClassic operation, please check it out. 

I will close this issue for now, please feel free to reopen or fire a new one if you find any issues.

Thanks!

closed 2018-10-06 by whitedogg13