revtel/react-native-nfc-manager #56: TypeError undefined is not an object

vivikenen, 2018-05-03:
Hi there, 
I was trying to implement this package on my react-native project, but when I try to reference the functions from the `NFCManager` from `react-native-nfc-manager`, all the functions catch error instead of going to `.then`.

`"react-native": "^0.47.0",`:
`"react": "^16.0.0-alpha.12",`

Android Device:
`Android Version: 7.1.1`

I did the following setting steps: 
`npm install -s react-native-nfc-manager`
`react-native link react-native-nfc-manager`

I try to print its error message, it says:
[image]

vivikenen, 2018-05-03:
When I try to link the package, the response looks different from another package
[image]

vivikenen, 2018-05-03:
@whitedogg13  Thx for replying~
I'm not using the expo, should be `react-native-cli`

whitedogg13, 2018-05-03:
Hi, if you're trying manually import, you can refer to #33 

Thanks!

vivikenen, 2018-05-03:
@whitedogg13  thx for your replying, the package has been link successfully now~

closed 2018-05-03 by vivikenen