revtel/react-native-nfc-manager #60: NFC Scan Sounds

dspencer926, 2018-06-11:
Hi, I'm trying to disable the native scan sound so that I can use my own instead.  Is there any way to use the enableReaderMode API instead of enableForegroundDispatch so that I can use the FLAG_READER_NO_PLATFORM_SOUNDS flag?

whitedogg13, 2018-06-11:
Hi, currently we haven't bridged the `enableReaderMode` API. Welcome to submit a PR!

whitedogg13, 2019-02-16:
@dspencer926 Hi, with the latest 1.2.0 release, now we can pass this flag with `registerTagEvent` to disable the sound!

closed 2019-02-16 by whitedogg13

JohnTheo13, 2019-08-23:
Which flag is this cause it's not mentioned in the documentation

rahamin1, 2020-06-30:
I assume that somebody knows which flag it is...

belgiancoder, 2020-09-02:
@rahamin1, I see that you haven't opened a new issue. Could you share your solution if you've solved it ?

For future reference, I will link [this part](https://github.com/cts-yazilim/react-native-nfc-manager#registertageventlistener-alertmessage-invalidateafterfirstread) of the documentation. Not because it's related to the current issue, but because it shows the correct syntax for adding and composing tags.

rahamin1, 2020-09-02:
I didn't solve it.

You can see some more info about this in https://stackoverflow.com/questions/62849096/disable-sound-in-react-native-nfc-manager-v2.