revtel/react-native-nfc-manager #63: iOS saying NFC not supported

rgpass, 2018-07-02:
## Issue

I have installed react-native-nfc-manager (RNNM for short) in my app and it works perfectly on Android but not on my iOS 8.

Note: This is my first time using a `react-native link`'ed module.

## Code

```js
NfcManager.start({
  onSessionClosedIOS: () => {
    console.log('ios session closed')
  },
})
  .then(result => {
    console.warn('start OK', result)
  })
  .catch(error => {
    console.warn('device does not support nfc!')
  })

NfcManager.registerTagEvent(
  tag => {
    // console.warn('Tag Discovered', tag)
    // console.warn(JSON.stringify(tag, null, 2))
    // console.warn(tag.ndefMessage[0])
    console.warn(NdefParser.parseText(tag.ndefMessage[0]))
  },
  'Hold your device over the tag',
  true
)
```

## Current Behavior

*Android*

Warning of `start OK undefined`. When scanning an NDEF NFC card, I get the message that's encoded. In other words, the line of `console.warn(NdefParser.parseText(tag.ndefMessage[0]))` prints out the message I want.

*iOS*

Warning of `device does not support nfc!`. Another warning of

```
Sending `NfcManagerSessionClosed` with no listeners registered.
```

## Expected Behavior

Both to behavior like Android's current behavior listed above.

## Theories

* I didn't do the Podfile installation process because I thought the `react-native link` would do it under-the-hood
* I didn't do anything regarding NFC permissions with Xcode

whitedogg13, 2018-07-03:
Hi, thanks for trying this library!

You will need to do some configuration for iOS, here you can find the details:
* https://github.com/hansemannn/iOS11-NFC-Example

rgpass, 2018-07-03:
That worked. Thanks! Good job on the library -- it works wonderfully!

whitedogg13, 2018-07-04:
I'm glad to help!

closed 2018-07-04 by whitedogg13