revtel/react-native-nfc-manager #64: Re-prompting on iOS

rgpass, 2018-07-09:
## Issue

I have the library installed and working on iOS. I have set a button to run `NfcManager.start()` which opens the iOS prompt to scan the device. When pressing the button the first time, the prompt shows up. When pressing cancel then the button a second time, nothing shows up.

## Code

```js
const handleOnPress = () => {
  console.warn('pressed')

  NfcManager.start()

  const { sendReading } = this.props
  NfcManager.registerTagEvent(
    tag => {
      const reading = NdefParser.parseText(tag.ndefMessage[0])
      sendReading(reading)
    },
    'Hold your device over the tag',
    true
  )
}
```

## Current Behavior

*iOS*

Prompt not showing up a second time when the button is tapped.

*Android*

Works perfectly. This is because it's an "always-on" state, rather than requiring a button to toggle a prompt.

## Expected Behavior

After tapping the button a second time, I expected the prompt to show up again.

## Theories

* Need to call `stop()` when it is closed and restart it. This didn't work. See code below.

```js
const handleOnPress = () => {
  console.warn('pressed')

  NfcManager.start({
    onSessionClosedIOS: () => {
      NfcManager.stop().then(() => handleOnPress())
    },
  })

  const { sendReading } = this.props
  NfcManager.registerTagEvent(
    tag => {
      const reading = NdefParser.parseText(tag.ndefMessage[0])
      sendReading(reading)
    },
    'Hold your device over the tag',
    true
  )
}
```

whitedogg13, 2018-07-09:
Hi, in above code try to use `unregisterTagEvent` rather than `stop` (in `onSessionClosedIOS`) should work!

rgpass, 2018-07-10:
Got it working! It may be good to put this in the `README.md`. My example that I got working:

```js
constructor(props) {
  super(props)

  this.handleOnPress = this.handleOnPress.bind(this)
}

componentWillMount() {
  this.handleOnPress()
}

handleOnPress() {
  NfcManager.start({
    onSessionClosedIOS: () => {
      NfcManager.unregisterTagEvent()
        // Throws error but still works
        // This silences the error
        .catch(() => {})
    },
  })

  const { handleSendReading } = this.props
  NfcManager.registerTagEvent(
    tag => {
      const reading = NdefParser.parseText(tag.ndefMessage[0])
      handleSendReading(reading)
    },
    'Hold your device over the tag',
    true
  )
}
```

Thanks again for the help.

closed 2018-07-12 by rgpass

rahamin1, 2020-09-23:
@rgpass Long time passed, I hope that you remember...

I assume that after calling NfcManager.unregisterTagEvent() in onSessionClosedIOS, you also call NfcManager.registerTagEvent, right?

rgpass, 2020-09-23:
Sorry dude, I do not remember. Best of luck, though!