revtel/react-native-nfc-manager #65: API to get UID

rgpass, 2018-07-12:
## Issue

I can't access the meta info for the tag, such as UID.

## Code

```js
NfcManager.registerTagEvent(
  tag => {
    console.warn(tag)
  },
  'Hold your device over the tag',
  true
)
```

## Current Behavior

*iOS & Android*

The `tag` returned does not include any information about the id. `tag.ndefMessage[0].id` exists but is an empty array.

## Expected Behavior

Either `tag.ndefMessage[0].id` returns the UID or `tag.uid` returns the UID.

## Theories

None

whitedogg13, 2018-07-16:
Hi, sorry for the late response!

For iOS, there's no UID available.
For Android, it should be in `tag.id` rather than `tag.ndefMessage[0].id`. The former one is uid and the later one is corresponding to the Ndef Record id which is rarely used.

Thanks!

rgpass, 2018-07-17:
Gotcha. Is it not possible on iOS at all or just with this library?

rgpass, 2018-07-18:
I see. Hopefully this changes in the future!

closed 2018-07-18 by rgpass

cavasinf, 2024-01-18:
> [!NOTE]  
> In order not to "scare" new developers, I'll write it there.

Using `NDEF` technology you can get the UID on **iOS AND Android** like this:
```js
try {
  await NfcManager.requestTechnology([NfcTech.Ndef])
  const tag = await NfcManager.getTag()
} catch (ex) {
  console.info(ex)
} finally {
  await NfcManager.cancelTechnologyRequest()
}

console.log(tag.id) // Return the UID
```

pke, 2024-04-24:
@cavasinf This also works with ISO tech on iOS