revtel/react-native-nfc-manager #66: Is it possible to make a nfc readOnly?

MatheusNaldi, 2018-07-13:
I saw this on android docs: https://developer.android.com/reference/android/nfc/tech/Ndef#makeReadOnly()

Is it possible to make a NFC tag readOnly with this package?

whitedogg13, 2018-07-16:
Yes, sorry for the late response!

This library currently only support making tags read-only with ndef write (because normally we freeze a tag after writing some new stuff to it) If it's you use case, you can passing `formatReadOnly` as true when calling `requestNdefWrite` API to make it happens, like this:

```javascript
NfcManager.requestNdefWrite(null, {formatReadOnly: true})
    .then(() => console.log('done!'))
    .catch(err => console.warn(err))
```

For the case without ndef writing, I can bridge the `makeReadOnly` API into this library later, and response to this thread when it's done.

Thanks!

whitedogg13, 2018-07-21:
Hi, I have added a new method `makeReadOnly` into latest release (0.5.2) which can force this behavior. Here's an example:
```javascript
NfcManager.registerTagEvent(tag => console.log(tag))
    .then(() => NfcManager.requestTechnology(NfcTech.Ndef))
    .then(() => NfcManager.makeReadOnly())
    .then(result => console.log('makeReadOnly result ', result))
```

Please use latest release to try it, thanks!

MatheusNaldi, 2018-07-23:
Hi, sorry for the late response, I was doing other tasks first.

Thanks for your time on this, I will update and test the makeReadOnly function today!

MatheusNaldi, 2018-07-24:
It worked, thanks!!!

I just have one question, I only work with Ndef tags, is it really required to ask for technology every time? It isn't better to just throw any error and let the application handle it? Or it isn't possible?

But this is not a problem, it's just a question, thanks for solving my problem!

closed 2018-07-24 by MatheusNaldi

whitedogg13, 2018-07-24:
Hi @MatheusNaldi , it's good to hear that your app works! 
Regarding you question, in Android simply reading NDEF (which used by most users) and write / make readonly are using different APIs, and the later one requires more setup such as casting tag handle to specific tech type. So in short, what you mentioned is possible, but I'd like to keep the way Android provides us for now~