revtel/react-native-nfc-manager #67: Is It Possible To Make A Read and Write Action at Once ?

VincentAlexander, 2018-07-17:
Is it possible to make a read and write action for only once NFC tapping?
For example I need to get the text from the NFC Card and then write it back to the NFC Card.

I can do it but I need twice tapping, the first tap will get or read the text from the NFC Card, and then the second tap will write the text to the NFC Card. 

So I need to do that two actions in one NFC tap.
Can someone help me with this? Thank you.

whitedogg13, 2018-07-18:
Hi, it's possible! 

It can be achieved by using `NfcTech`, the code will look like this:
```javascript
NfcManager.registerTagEvent(tag => console.log(tag)) // first step will always be registering tags
    .then(() => NfcManager.requestTechnology(NfcTech.Ndef)) // request raw NFC tech handler
    .then(() => NfcManager.getNdefMessage()) // success, read the original data
    .then(tag => { // the original data is resolved successfully
        let parsedText = parseText(tag); 
        this.setState({tag, parsedText})
    })
    .then(() => NfcManager.writeNdefMessage(buildTextPayload(textToWrite))) // write new data
    .then(cleanUp)
```

You can see the `example/AndroidTechTestNdef.js` for a full demo.

VincentAlexander, 2018-07-19:
Oh, okay. I'll try it later, thank you so much!

closed 2018-08-02 by whitedogg13

pruthvirajkarurcodewave, 2022-09-15:
Is It Possible To Make A Read and Write Action at Once in iOS as well ?