revtel/react-native-nfc-manager #69: Everything fails

Jturpijn, 2018-07-20:
I created a react-native app which I then used it with expo.io, after finishing the basic design of the app I detached in order to use this library. Everything kinda works, I don't receive any errors, but everything fails.  
  
I receive two warnings, the first one being a promise rejection:
```
[10:41:53] [Unhandled promise rejection: TypeError: undefined is not an object (evaluating 'NativeNfcManager.isSupported')]
- node_modules\react-native-nfc-manager\NfcManager.js:60:23 in <unknown>
- node_modules\promise\setimmediate\core.js:45:7 in tryCallTwo
- node_modules\promise\setimmediate\core.js:200:23 in doResolve
- node_modules\promise\setimmediate\core.js:66:12 in Promise
- node_modules\react-native-nfc-manager\NfcManager.js:59:23 in isSupported
* screens\NfcScreen.js:56:8 in componentDidMount
- node_modules\react-proxy\modules\createPrototypeProxy.js:61:45 in proxiedComponentDidMount
- node_modules\react-native\Libraries\Renderer\ReactNativeRenderer-dev.js:10627:12 in commitLifeCycles
- ... 17 more stack frames from framework internals
```

And the second one being a registerTagEvent fail: 
```
[10:47:20] registerTagEvent fail
- node_modules\react-native\Libraries\ReactNative\YellowBox.js:80:15 in warn
- node_modules\expo\src\Expo.js:25:41 in warn
* screens\NfcScreen.js:267:29 in <unknown>
- node_modules\promise\setimmediate\core.js:37:14 in tryCallOne
- node_modules\promise\setimmediate\core.js:123:25 in <unknown>
- ... 10 more stack frames from framework internals
```
Besides those warnings, the example app code doesn't recognize my NFC enabled, which I couldn't find the code for either way, so I'm not sure if that has anything to do with it not working. And I get this warning after I press 'stop Tag Detection'. : 
```
[10:23:01] unregisterTagEvent OK undefined
```

Now I'm not sure if these are caused by the tags, my Nexus 5x or the fact that I missed something. 
And I made sure that I linked the NFC manager to my react-native. I also looked at the example/app.js and all the issues that were submitted, but I'm just a junior trying my hand at NFC. Thanks in advance for any help! 

These are my dependencies:
```
    "@expo/samples": "2.1.1",
    "expo": "^28.0.0",
    "react": "16.3.1",
    "react-native": "github:expo/react-native#sdk-28.0.0",
    "react-native-nfc-manager": "^0.5.1",
    "react-navigation": "2.3.1"
```

Jturpijn, 2018-07-22:
Hey @whitedogg13 I figured something as such, but couldn't find what I did wrong. 
I first started my project through the [expo getting started](https://expo.io/learn)  
I wrote a basic app and then [detached from Expo](https://docs.expo.io/versions/v28.0.0/expokit/detach#__next) to use this third party library.   
I then followed the instructions from the nfc-manager readme to use the react-native link command. When using the command again I get this as a response : 
```
$ react-native link react-native-nfc-manager
Scanning folders for symlinks in X:\Ximedes\FP 5 Mobile\FP5\node_modules (40ms)
rnpm-install info Platform 'android' module react-native-nfc-manager is already linked
```

whitedogg13, 2018-07-22:
Hi, I think your step is correct.

One more question, how do you start your app after linking? You will need to do `react-native run-android`, this command will compile the project again including the native part. (If you simply restart your app and rely on RN packager, that won't work because now you have native code changes rather than JS code changes!)

Jturpijn, 2018-07-23:
Good to hear that I've done the right thing, unfortunately you're right about the starting command. I was still using the ```exp start``` command to start the project. I've been trying to use ```react-native run-android``` , but I've been running into this error : 

```
Building and installing the app on the device (cd android && gradlew.bat installDebug)...
Could not install the app on the device, read the error above for details.
``` 

But I'm fairly sure that this isn't related to your library. When I fixed that I will update the progression.

54vanya, 2018-07-27:
You should eject your project from Expo to use any of native modules. As I know Expo still doesn't have NFC support

whitedogg13, 2019-01-19:
Close it for now, please feel free to reopen if there's any update.

closed 2019-01-19 by whitedogg13

codelover1110, 2020-03-08:
dfd

codelover1110, 2020-03-08:
Hi. 
Can you help me?