revtel/react-native-nfc-manager #71: I get the payload data but cannot parse out the text. Please help,

themakerman, 2018-07-24:
`{ndefMessage: Array(1), maxSize: 868, type: "NFC Forum Type 2", techTypes: Array(3), id: "04F361429E4A80"}
id
:
"04F361429E4A80"
maxSize
:
868
ndefMessage
:
Array(1)
0
:
id
:
""
payload
:
Array(7)
0
:
87
1
:
97
2
:
115
3
:
32
4
:
117
5
:
112
6
:
10
length
:
7
__proto__
:
Array(0)
tnf
:
2
type
:
[]
__proto__
:
Object
length
:
1
__proto__
:
Array(0)
techTypes
:
(3) ["android.nfc.tech.NfcA", "android.nfc.tech.MifareUltralight", "android.nfc.tech.Ndef"]
type
:
"NFC Forum Type 2"`

whitedogg13, 2018-07-25:
Hi, sorry for the late response.

From the screenshot, your ndef's `tnf` value is 2, which refers to TNF_MIME_MEDIA, and currently ndef associated with this `tnf` not supported by our NdefParser API yet.

I will try to add this feature when I'm available and keep you posted!

themakerman, 2018-08-02:
no probz....great library mate

whitedogg13, 2018-08-03:
Hi, just perform some quick survey regarding this.

Fully support TNF_MIME_MEDIA parsing isn't an easy task, so it might need to use a external library such as [node-mime](https://github.com/broofa/node-mime).

However for your particular case, it might be easier to simply use `fromCharCode` to transform the payload back to text. For example:
```
text = tag.ndefMessage[0].payload.reduce(
  (acc, byte) => acc + String.fromCharCode(byte), ''
)
// text will be `Was up\n'
```

Hope that helps!

themakerman, 2018-08-10:
hmm thanks i'll try it out

whitedogg13, 2018-10-21:
close it for now due to no further activity, please reopen if the issue persists!

closed 2018-10-21 by whitedogg13