revtel/react-native-nfc-manager #72: NFC starts but no tag events [android]

JoeDurand3, 2018-07-29:
Hello,

NFC initializes just fine, not getting an error when calling start(), but the listener function is never called.

How can I diagnose where there might be a problem? I've tried lots of combinations of permissions and using the Android Device Monitor.

I think this a linking issue. I'm using react-native-navigation which seems to give others problems, along with firebase and several other libraries.

Here is my MainApplication.java:

```
package com.reactapp;

import android.app.Application;
import co.apptailor.googlesignin.RNGoogleSigninPackage;
import com.airbnb.android.react.maps.MapsPackage;
import com.facebook.appevents.AppEventsLogger;
import com.facebook.CallbackManager;
import com.facebook.FacebookSdk;
import com.facebook.react.ReactApplication;
import com.facebook.react.ReactNativeHost;
import com.facebook.react.ReactPackage;
import com.facebook.react.shell.MainReactPackage;
import com.facebook.reactnative.androidsdk.FBSDKPackage;
import com.facebook.soloader.SoLoader;
import com.reactnativenavigation.NavigationApplication;
import com.RNProximity.RNProximityPackage;
import community.revteltech.nfc.NfcManagerPackage;
import io.invertase.firebase.RNFirebasePackage;
import io.invertase.firebase.analytics.RNFirebaseAnalyticsPackage;
import io.invertase.firebase.auth.RNFirebaseAuthPackage;
import io.invertase.firebase.firestore.RNFirebaseFirestorePackage;
import java.util.Arrays;
import java.util.List;

public class MainApplication extends NavigationApplication implements ReactApplication {

  private static CallbackManager mCallbackManager = CallbackManager.Factory.create();

  protected static CallbackManager getCallbackManager() {
    return mCallbackManager;
  }

  private final ReactNativeHost mReactNativeHost = new ReactNativeHost(this) {
    @Override
    public boolean getUseDeveloperSupport() {
      return BuildConfig.DEBUG;
    }

    @Override
    protected List<ReactPackage> getPackages() {
      return Arrays.<ReactPackage>asList(
          new MainReactPackage(),
            new FBSDKPackage(mCallbackManager),
            new NfcManagerPackage(),
            new MapsPackage(),
            new RNFirebaseAnalyticsPackage(),
            new RNFirebaseAuthPackage(),
            new RNFirebaseFirestorePackage(),
            new RNFirebasePackage(),
            new RNGoogleSigninPackage(),
            new RNProximityPackage()            
      );
    }

    @Override
    protected String getJSMainModuleName() {
      return "index";
    }
  };

  @Override
  public List<ReactPackage> createAdditionalReactPackages() {
    return Arrays.<ReactPackage>asList(
          new MainReactPackage(),
            new FBSDKPackage(mCallbackManager),
            new NfcManagerPackage(),
            new MapsPackage(),
            new RNFirebasePackage(),
            new RNFirebaseAnalyticsPackage(),
            new RNFirebaseAuthPackage(),
            new RNFirebaseFirestorePackage(),
            new RNGoogleSigninPackage(),
            new RNProximityPackage()
      );
  }

  @Override
  public ReactNativeHost getReactNativeHost() {
    return mReactNativeHost;
  }

  @Override
  public boolean isDebug() {
      return BuildConfig.DEBUG;
  }

  @Override
  public void onCreate() {
    super.onCreate();
    FacebookSdk.sdkInitialize(getApplicationContext());
    SoLoader.init(this, /* native exopackage */ false);
  }
}

```

whitedogg13, 2018-08-02:
Hi, @JoeDurand3 sorry for the late response.

We do have users reporting that there will be link issue when combining this lib with `react-native-navigation`, in #33 . However, from the code above I  believe you have properly linked the library. 

Can you attach some native log for further analyze?

JoeDurand3, 2018-08-02:
Hi @whitedogg13 , no problem at all. I appreciate you looking into this with me.

Attached is a log file from ADM. I reload the app and tap the tag several times.

Thanks
Joe
[joedurand3_log.txt](https://github.com/whitedogg13/react-native-nfc-manager/files/2255416/joedurand3_log.txt)

JoeDurand3, 2018-08-04:
Not that this is a solution - but I was able to quickly migrate my project over to react-navigation instead of react-native-navigation.

It's the superior library IMO, as it doesn't require any modifications to native MainApplication or AppDelegate. The API is simpler and, of course, does not interfere with NFC or other device APIs that RNN is notorious for breaking judging by their issue history.

whitedogg13, 2018-08-05:
Hi @JoeDurand3 thanks for keep digging this!

Thanks to the log you attached, I think I can spot the root cause.

First we observe the log, we can see:

* NFC event did fired by Android OS. The `NDEF_DISCOVERED` intent has been sent multiple times in the log.
* However, this intent is then somehow be forced with flag `FLAG_ACTIVITY_NEW_TASK`

To be more precise, I list partial log below (line 593, line 598):
```
08-02 17:53:56.906: I/ActivityManager(3927): START u0 {act=android.nfc.action.NDEF_DISCOVERED typ=text/plain flg=0x24000000 cmp=ComponentInfo{com.companyname.reactapp/com.co
...
08-02 17:53:56.913: W/ActivityManager(3927): startActivity called from non-Activity context; forcing Intent.FLAG_ACTIVITY_NEW_TASK for: Intent { act=android.nfc.action.NDEF_
```

Why this cause issues? Let me explain this in more detail:

* In Android NFC API, normally we require the same Activity to register and listener to NFC intent. It's well documented in [Android doc](https://developer.android.com/guide/topics/connectivity/nfc/advanced-nfc#foreground-dispatch), and the basic idea is:
    * set intent to be `FLAG_ACTIVITY_SINGLE_TOP` 
    * listen to new intent in the `onNewIntent` method in the same Activity
* However, since now the activity has been forced to `FLAG_ACTIVITY_NEW_TASK` (which means the intent will be sent to a newly created Activity, not the original one), so our Activity can no-longer receive the NFC intent

That's it for now :). I will try to adapt RNN with my lib some other day to see whether there's a way to avoid this!

Burnsy, 2018-08-08:
My team member also reported an issue, https://github.com/whitedogg13/react-native-nfc-manager/issues/74 but it could be linked to RNN as we use this as well. That is on IOS.

Thanks

whitedogg13, 2019-01-19:
Close it for now, please feel free to reopen if there's any update.

closed 2019-01-19 by whitedogg13