revtel/react-native-nfc-manager #76: Missing .podspec file

maciejimielowski, 2018-08-13:
While doing **Install with cocopods** part of setup and running `pod install` I got following message:

> No podspec found for 'react-native-nfc-manager' in '../node_modules/react-native-nfc-manager' 

While there is `react-native-nfc-manager.podspec` file in your repository, there is no such file in node_modules/react-native-nfc-manager directory. Additionally it isn't included in files in package.json
```
  "files": [
    "NfcManager.js",
    "NdefParser.js",
    "index.d.ts",
    "android",
    "ios",
    "example"
  ],
```
Is running `pod install` not necessary?

whitedogg13, 2018-08-16:
Sorry for the late response, it's indeed missing and I will add it back. Thanks for spotting this issue!

jlkalberer, 2018-08-22:
Any update on this or a workaround?

whitedogg13, 2018-08-23:
Hi, sorry for the delay. I have just added the podspec back to released npm module in latest v0.5.5 release. Thanks!

closed 2018-10-03 by whitedogg13