revtel/react-native-nfc-manager #78: [Question] How to organize P2P data exchange between 2 custom applications?

paveldz, 2018-08-22:
Hi!
I'm trying to figuring out how to organize P2P data exchange using this library.
What I would like to get: two custom apps that can exchange data using NFC.

Workflow: 
1) App1 initiates data exchange session.
2) App2 reads data of App1, does some work and then sends response back to App1.

**Is this achievable using this library? If yes, what steps should I follow?
If this not achievable with this library, is this achievable at all?**

I thought `setNdefPushMessage(bytes)` method will help me, but looks like it's for Android Beam only, not for custom apps, right?
Possible, `NfcTech.NfcA` functionality and `transceive` method can be helpful for my task?

whitedogg13, 2018-08-23:
Hi, currently no public API can accomplish the feature you're asking for. 

The reason for that is Android NFC `P2P Mode` API only allow one directional data transfer for now (the underline OS can actually do bi-directional stuff but the API isn't exposed to public).

However, the Android NFC `Card Emulation` mode API (ie, `HCE` for Host Card Emulation) can perform bi-directional data transfer, but this library doesn't bridge those APIs.

paveldz, 2018-08-23:
Supporting of `Card Emulation` would be very helpful.

As I understand `NfcTech.NfcA` functionality and `transceive` method can be used to push command to tag and get a response back, right? Can another android device be in tag role that can accept command and push response back?

whitedogg13, 2018-08-27:
Hi, sorry for the late response.

You're correct about `transceive`, this API can indeed be used to send/receive payload between `NFC tag` and `NFC reader`. However it still cannot achieve bi-directional data transfer because the Android device normally can only act as `NFC reader` rather than `NFC tag`.

Thanks!

closed 2018-08-28 by paveldz