revtel/react-native-nfc-manager #81: Write NDEF Array

ubaycreative, 2018-09-06:
Hi, I'm trying to write ndef array more than 1 index
this `buildTextPayload` is really cool, but how can i change this to ndef array?

`function buildTextPayload(valueToWrite) {
    const textBytes = strToBytes(valueToWrite);
    // in this example. we always use `en`
    const headerBytes = [0xD1, 0x01, (textBytes.length + 3), 0x54, 0x02, 0x65, 0x6e];
    return [...headerBytes, ...textBytes];
}`

whitedogg13, 2018-09-09:
Hi, sorry for the late response.

Currently we're working on a feature to enhance NDEF message building & parsing, which includes the scenario you mentioned. So this issue will be resolved once we complete the feature.

We hope to deliver it soon and thanks for the advice!

whitedogg13, 2018-09-17:
Hi @ubaycreative , we have enabled this feature in the latest 0.6.0 release (also include a full demo in example/MultiNdefRecord.js). Please check it out, thanks!

closed 2018-10-03 by whitedogg13

ubaycreative, 2018-10-19:
@whitedogg13 it's cool thanks for the update