revtel/react-native-nfc-manager #82: Couldn't detect NFC reader (ACR122)

gustavomanolo, 2018-09-10:
Hi, i've connected an NFC reader directly to my Android device (my android device doesn't support native NFC), but i believe the library doesn't detect NFC readers connected though USB, do you know how to detect and work with this?
The model of the NFC reader that i've connected is:
Brand: ACS
Model: ACR122
P/N: ACR122U-A9
Power: 5VDC, 200mA

Thanks a lot for your time.

whitedogg13, 2018-09-12:
Hi, sorry for the late response.

This project is not meant to support an external connected NFC reader. Alternatively, it bridges device's (iOS or Android) native NFC API to React Native side.

Thanks!

gustavomanolo, 2018-09-18:
@whitedogg13 Thanks for your response.

closed 2018-09-18 by gustavomanolo

andrewasfa, 2019-02-22:
> Hi, i've connected an NFC reader directly to my Android device (my android device doesn't support native NFC), but i believe the library doesn't detect NFC readers connected though USB, do you know [...]

Hey sorry for digging this out. But did you find a solution for using React-Native with this external reader?

whitedogg13, 2019-02-23:
Hi, @andrewasfa I have never investigated this topic, so no further info can be provided.

andrewasfa, 2019-02-25:
> Hi, @andrewasfa I have never investigated this topic, so no further info can be provided.

Thank you for your response @whitedogg13 
Sorry for spamming this thread, but since I don't think I can message directly on Github, @gustavomanolo did you have any success? If so, there is a question on StackOverflow which would appreciate your answer :) https://stackoverflow.com/questions/54830063/has-anyone-got-usb-nfc-reader-acr122u-to-work-with-react-native