revtel/react-native-nfc-manager #84: how to write android package type

ibnumuntoha, 2018-09-18:
if write type text is

`function buildTextPayload(valueToWrite) {
    const textBytes = strToBytes(valueToWrite);
    // in this example. we always use `en`
    const headerBytes = [0xD1, 0x01, (textBytes.length + 3), 0x54, 0x02, 0x65, 0x6e];
    return [...headerBytes, ...textBytes];
}`
how to write nfc tag type android package, so when i scan nfc tag from my device going to the apps
and how to write more than one data .
first data is text type, second is android package

whitedogg13, 2018-09-18:
Hi, you can try the following example:

```javascript
import NfcManager, {Ndef} from 'react-native-nfc-manager'
...
let bytes = Ndef.encodeMessage([
    Ndef.textRecord("hello, world"),
    Ndef.androidApplicationRecord("my-app-pkgname"),
]);

NfcManager.requestNdefWrite(bytes)
    .then(() => console.log('write completed'))
```

Thanks!

ibnumuntoha, 2018-09-19:
hi, it's worked ,
thank you for your help

closed 2018-10-03 by whitedogg13