revtel/react-native-nfc-manager #85: NfcV

krokhale, 2018-09-19:
Thanks in advance for reading this!

I am running an experiment and these are some of my observations:

- I am trying to scan a medical device attached to the body that detects blood glucose levels for diabetics.
- You should find a lot of relevant info that might be useful here: https://www.reddit.com/r/diabetes/comments/2ld5t5/freestyle_libre_used_nfc_sensor_hack/

- I got your example up and running and scanned the sensor, and this is what i got, i am trying to make sense of what i could do further. I was expecting an array of tags in the data. Am i missing something. Or is the data being read off the tag not parseable. If it's not, can i get access to the raw data stream and play with the hex/binary stream to try to decode it? If so what's the best way to start?

Any pointers will be appreciated!

[image]

whitedogg13, 2018-09-21:
Hi, sorry for the late response.

It seems that the tag doesn't contain any NdefMessages. This library by default will only read the NdefMessage from tag, so there won't be any data shown.

Android does support direct comm with NfcV tag (primarily through a `transceive` API), as can be seen here: 
* https://developer.android.com/reference/android/nfc/tech/NfcV

Currently this library doesn't bridge this API yet, I will try to implement this when available.

Thanks

krokhale, 2018-09-21:
Thanks, i think i can try and implement that. 

Can you point me to the relevant part in your code that i can start looking at. Appreciate the help!

whitedogg13, 2018-10-21:
Hi, @krokhale , we just release v1.0.0 which supports NfcV `transceive`, please take a look!