revtel/react-native-nfc-manager #87: Is There A Limit Character for requestNdefWrite ?

VincentAlexander, 2018-10-02:
Hi,

I'm trying to insert a long character maybe 300 character to NFC Card (Classic 1K size).
But my apps always show error "Incorrect ndef format".
And I think the error came from this code:
`NfcManager.requestNdefWrite(bytes)
    .then(() => console.log('write completed'))
    .catch(err => console.warn(err))`

When I adjust the character to 100 character long it works successfully, but when I made it to 300 character it goes error. Can someone help me with this?

Thank you,

Vincent

whitedogg13, 2018-10-03:
Hi, can you share how you build your ndef message bytes? let's see if the logic is correct.

VincentAlexander, 2018-10-03:
Hi,

I've searched for the character limitation and I found that the character limit for requestNdefWrite is 253 character. If goes above that for example 254 character it will goes "Incorrect ndef format" error.

This is the function genSPBString() I use to generate my NFC string.
> genSPBString = () => {
        var newSPB = [
            this.state.idKartu,
            this.state.idSpb,
            this.state.nomorSPB,
            this.state.truck_code,
            parseInt(this.state.jumlah_tandan,10),
            parseFloat(this.state.jumlah_brondolan,10),
            parseInt(this.state.jumlah_OPH,10),
            this.state.tujuan,
            this.state.supir,
            parseFloat(this.state.estimasi_tonase,10),
            this.state.tanggal,
            this.state.stateCurrentClock,
            this.state.employeeCode,
            "testing123123",
            "testing123123",
            "testing123123",
            "testing123123",
            "testing123123",
            "testing123123",
            "testing123123",
            "testing123123",
            "testing123123",
            "testing123123",
            "testing12"
        ]
        console.log(newSPB)
        return newSPB.join()
    }

The console.log(newSPB) will get this result. In my case this string limited to 253 character.
> S001,2120280309491934M,001,B8899UAF,0,0,0,INTP2,SC/INIT1/0106/66,0,03/10/2018,09:49:19,SC/INIT1/0107/35,testing123123,testing123123,testing123123,testing123123,testing123123,testing123123,testing123123,testing123123,testing123123,testing123123,testing12345

Then I pass the newSPB variable to urlToWrite state.
> this.setState({
     urlToWrite: encryptedSpb
}

And then I call the requestNdefWrite() function.
In there future the generated NFC string from genSPBString() will also encrypted using AES Encryption using this library https://github.com/rocwangv/react-native-aes-kit#readme. This is the result for the encryption.
> fn7ii0WGCX9pZ530V3tK1snoiUSoIg2KdGLSc7MqNbLuuC6QU2j7hsmCbNm1mrUNtVxPxswNsE7C
vdj094DLNbfbBSLpFXhlMiC+ZJ8i9QZ6L9nkwF3TWgz+HFOxt1YmKryjJRykAVdWMbCd4GFrxNj7
q8yGzHPSqrCZwXK6LqBmTxCDNr6jB0Vtqz1l/googf84egHtNTnGuRytTnendPjEJWNvTwG4TzoQ
jwciZU1FB1JC1jCb6ghdu6Or9Y58r4fC+9Ac6jHlw+Z+62kdcCDddCzy5OhmTiokDnvJyxBh6PVX
WMyj97wRS69NEz8xoZpNoaeeJRsXkbO6MBBbbg==

Note:
I think my logic to generate the NDEF message are already correct, because like I said when I make the string to 100 character long, it successfully write to the NFC card with the encryption string. When I read the card I do the decryption first then read the variables, it goes success without error before I make the genSPBString() character long to 300++.

whitedogg13, 2018-10-03:
Hi, thanks for responding, however what I need to know is how you construct your Ndef Message `bytes` from the url string, so I can see whether you build the Ndef Message payload correctly.

More specifically, how you get from your `urlToWrite` to final `bytes`? (also, some old example I provided might not able to encode correct Ndef Message payload...)

VincentAlexander, 2018-10-03:
Hi,

Hmm I think to convert urlToWrite into bytes I'm using the same like this library. Here's the function. 

> const RtdType = {
    URL: 0,
    TEXT: 1
};

>         _NfcRequestNdefWrite = () => {
            function NfcStrToBytes(str) {
                let result = [];
                for (let i = 0; i < str.length; i++) {
                    result.push(str.charCodeAt(i));
                }
                return result;
            }
    
            function NfcBuildUrlPayload(valueToWrite) {
                const urlBytes = NfcStrToBytes(valueToWrite);
                // in this example, we always use `http://` 
                const headerBytes = [0xD1, 0x01, (urlBytes.length + 1), 0x55, 0x03]; 
                return [...headerBytes, ...urlBytes];
            }
    
            function NfcBuildTextPayload(valueToWrite) {
                const textBytes = NfcStrToBytes(valueToWrite);
                // in this example. we always use `en`
                const headerBytes = [0xD1, 0x01, (textBytes.length + 3), 0x54, 0x02, 0x65, 0x6e];
                return [...headerBytes, ...textBytes];
            }
    
            let {isWriting, urlToWrite, rtdType} = this.state;
            if (isWriting) {
                return;
            }
    
            let bytes;
    
            if (rtdType === RtdType.URL) {
                bytes = NfcBuildUrlPayload(urlToWrite);
            } else if (rtdType === RtdType.TEXT) {
                bytes = NfcBuildTextPayload(urlToWrite);
            }

            console.log("BYTES = " + bytes)
    
            this.setState({isWriting: true});
            NfcManager.requestNdefWrite(bytes)
                .then(() => this.setState({suksesWrite: true}))
                .catch(err => {
                    this.setState({
                        suksesWrite: false
                    })
                    console.warn(err)
                    this._hideModal()
                    this._NfcStopDetection()
                    alert(err)
                    // toastError("Proses NFC gagal, silahkan coba lagi!")
                })
                .then(() => this.setState({isWriting: false}))
        }

I'm sorry if I give the wrong answer. Is this what you need to know? I really appreciate your reply, thank you so far.

whitedogg13, 2018-10-03:
Cool, that's indeed what I'm looking for. Looks like you're using old code snippet to encode NDEF payload, which might cause issue like this.

Please update to the latest version, and try following code:
```
import NfcManager, {Ndef} from 'react-native-nfc-manager'; 

function buildUrlPayload(valueToWrite) {
    return Ndef.encodeMessage([
        Ndef.uriRecord(valueToWrite),
    ]);
}
```

Thanks!

VincentAlexander, 2018-10-04:
Great it works! The encryption and decryption text also works perfectly.

Thank you so much!

closed 2018-10-04 by VincentAlexander