revtel/react-native-nfc-manager #88: getLaunchTagEvent() always returns none.

maxmijn, 2018-10-02:
Hi there,

I have a nfc tag that contains some plain text and the package name of my app. If I scan the tag, it successfully launches the app, but the `getLaunchTagEvent()` returns `null` every time. So whether I am manually opening the app or opening the app by scanning, the function always returns `null`. 

For now, I have to remove my phone from the tag and tap it again to retrieve the data. Is it possible to read this data right away?

whitedogg13, 2018-10-03:
Hi, thanks a lot for reporting. I will check soon and feedback soon.

whitedogg13, 2018-10-09:
Hi, sorry for the late response. 

I have just tested this feature, however it's working as expected. Actually, this feature is pretty straightforward, it simply call getIntent in the MainActivity, and send it back to JS side. 

In order to further analyze this, please help to provide following info:

* Tag's NDEF type (the one you use to launch app, is it Android Application Record? or URI record? )
* Do you edit your AndroidManifest to include the proper intent filter to match your tag?
* Do you put extra Activity other than the MainActivity?
* Do you use native navigation solution such as RNN?

Thanks!

maxmijn, 2018-10-10:
Hi, thanks for your response!

To answer your questions: 

NDEF type is an AAR, so 'com.example'.

My manifest file looks like this:

```<uses-permission android:name="android.permission.INTERNET" />
    <uses-permission android:name="android.permission.SYSTEM_ALERT_WINDOW"/>
    <uses-permission android:name="android.permission.VIBRATE"/>
    <uses-permission android:name="android.permission.NFC"/>
    <uses-feature android:name="android.hardware.nfc" android:required="false" />

    <application
      android:name=".MainApplication"
      android:label="@string/app_name"
      android:icon="@mipmap/ic_launcher"
      android:allowBackup="false"
      android:theme="@style/AppTheme">
      <activity
        android:name=".MainActivity"
        android:label="@string/app_name"
        android:configChanges="keyboard|keyboardHidden|orientation|screenSize"
        android:screenOrientation="portrait"
        android:windowSoftInputMode="adjustPan">
        <intent-filter>
            <action android:name="android.intent.action.MAIN" />
            <category android:name="android.intent.category.LAUNCHER" />
        </intent-filter>
      </activity>
      <activity android:name="com.facebook.react.devsupport.DevSettingsActivity" />
    </application>
```
And I do not use React Native Navigation.

The app successfully launches on scanning the tag, and I can also successfully read tags. It is just that I do not receive the getLaunchTagEvent.

Thanks for helping!

maxmijn, 2018-10-10:
Hmm seem to have fixed it by adding the following intent filter to my Manifest file:

```        <intent-filter>
            <action android:name="android.nfc.action.NDEF_DISCOVERED"/>
            <category android:name="android.intent.category.DEFAULT"/>
            <data android:mimeType="text/plain" />
        </intent-filter>
```
The only problem now is that everytime the tag is scanned, the app relaunches.

whitedogg13, 2018-10-13:
Hi, sorry for the late response.

`everytime the tag is scanned, the app relaunches` --> not sure the exactly meaning here, but maybe you can try to change the `launchMode` for `MainActivity` in manifest to fit your need.

fabioloreggian, 2018-12-02:
> Hmm seem to have fixed it by adding the following intent filter to my Manifest file:
>
> ```
>         <intent-filter>
>             <action android:name="android.nfc.action.NDEF_DISCOVERED"/> [...]

Can you please provide your entire file for context please, because I am not getting the launch tag event working. The tag launches the app but my tag is null.

whitedogg13, 2018-12-09:
Hi @maxmijn sorry for the late response, kinda busy recently.

Per my understanding, the reason why your app is relaunched when tapping a new tag is related to your `launchMode` setting (in `AndroidManifest`, for the your `MainActivity`), not related to the library. By default, android allows the incoming intent to instantiate an activity more than once, I guess that's the reason why your app relaunches. You can try to set `launchMode` to `singleTop` or `singleTask` to see which one suites your need.

Thanks!

closed 2019-01-03 by whitedogg13

KinjalDroid, 2019-05-15:
App does not launch by default when nfc tag is scanned.. Should I change in code ? And also I am not getting the launch tag event working. The tag launches the app but my tag is null. 
Please help.. SO what should i add in manifest file ?
This is manifest file code. 

 `

    <uses-permission android:name="android.permission.INTERNET" />
    <uses-permission android:name="android.permission.SYSTEM_ALERT_WINDOW"/>
    <uses-permission android:name="android.permission.VIBRATE"/>
     <uses-permission android:name="android.permission.CAMERA" />
     <uses-permission android:name="android.permission.READ_EXTERNAL_STORAGE" />
<uses-permission android:name="android.permission.WRITE_EXTERNAL_STORAGE" />
<uses-permission android:name="android.permission.FLASHLIGHT" />
 <uses-permission android:name="android.permission.ACCESS_WIFI_STATE" />
    <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
 <uses-permission android:name="android.permission.NFC"/>
    <uses-feature android:name="android.hardware.nfc" android:required="false" />
     

    <application
      android:name=".MainApplication"
      android:label="@string/app_name"
      android:icon="@mipmap/ic_launcher"
      android:allowBackup="false"
      android:theme="@style/AppTheme">
      <activity
        android:name=".MainActivity"
        android:label="@string/app_name"
        android:configChanges="keyboard|keyboardHidden|orientation|screenSize"
        android:windowSoftInputMode="adjustResize">
        <intent-filter>
            <action android:name="android.intent.action.MAIN" />
            <category android:name="android.intent.category.LAUNCHER" />
        </intent-filter>
          <intent-filter>
            <action android:name="android.nfc.action.TECH_DISCOVERED"/>
            <category android:name="android.intent.category.DEFAULT"/>
            <data android:mimeType="text/plain" />
        </intent-filter>
      </activity>
      <activity android:name="com.facebook.react.devsupport.DevSettingsActivity" />
               
    </application>

`

Please correct me.

Atakspor, 2019-10-17:
> Can you please provide your entire file for context please, because I am not getting the launch tag event working. The tag launches the app but my tag is null.

Same here. I am able to launch the app with AAR but the return value from `getLaunchTagEvent()` is always null.