revtel/react-native-nfc-manager #90: Password protection support

pbuccigrossi, 2018-10-04:
It would be great to have password protection support to write and protect data

whitedogg13, 2018-10-06:
Hi @pbuccigrossi , as far as I know, those password protection features are mostly proprietary to tag vendors and not part of Android API, so we won't be able to provide them directly in this library. 

However, if you give us the concrete tag spec, maybe we can try to make a example. 

Thanks!

whitedogg13, 2018-10-21:
close it due to no further activity

closed 2018-10-21 by whitedogg13

KBog, 2020-01-12:
Any activity on the password protection support?

traxx10, 2020-05-09:
Its needed now

developer8530, 2023-12-18:
export const pack = [0x00, 0x00];
async function ensurePasswordProtection(value) {
  const _password = getBytes(value);
  const password = strToHex(value);
  let respBytes = null;
  let writeRespBytes = null;
  let authPageIdx;

  try {
    if (Platform.OS === "android") {
      getOutlet("androidPrompt").update({
        visible: true,
        message: "Ready to scan NFC",
      });
    }
    // check Tag Type
    // await NfcManager.requestTechnology(NfcTech.NfcA);
    await NfcManager.requestTechnology([NfcTech.NfcA, NfcTech.Ndef]);
    // let bytes = Ndef.encodeMessage([Ndef.textRecord(password, "en", "UTF8")]);
    // console.log("bytes::::::::", bytes);
    respBytes = await NfcManager.nfcAHandler.transceive([0x30, 0]);
    const cc2 = respBytes[14];
    //todo more conditions to be implemented
    if (cc2 === 109) {
      authPageIdx = 227; // NTAG 216
    } else if (cc2 === 62) {
      authPageIdx = 131; // NTAG 215
    } else if (cc2 === 18) {
      authPageIdx = 41; // NTAG 213
    } else if (cc2 === 16) {
      authPageIdx = 37; // NTAG 212
    } else if (cc2 === 6) {
      authPageIdx = 37; // ultralight ev1
    } else {
      authPageIdx = 16; // NTAG 210
    }

    // check if AUTH is enabled
    respBytes = await NfcManager.nfcAHandler.transceive([0x30, authPageIdx]);
    const auth = respBytes[3];
    if (auth === 255) {
      // configure the tag to support password protection
      writeRespBytes = await NfcManager.nfcAHandler.transceive([
        0xa2,
        authPageIdx + 3,
        ...pack,
        respBytes[14],
        respBytes[15],
      ]);
      writeRespBytes = await NfcManager.nfcAHandler.transceive([
        0xa2,
        authPageIdx + 2,
        ...password,
      ]);
      writeRespBytes = await NfcManager.nfcAHandler.transceive([
        0xa2,
        authPageIdx + 1,
        respBytes[4] & 0x7f,
        respBytes[5],
        respBytes[6],
        respBytes[7],
      ]);
      writeRespBytes = await NfcManager.nfcAHandler.transceive([
        0xa2,
        authPageIdx,
        respBytes[0],
        respBytes[1],
        respBytes[2],
        4,
      ]);
      if (Platform.OS === "ios") {
        await NfcManager?.setAlertMessageIOS("Success");
      }
      if (Platform.OS === "android") {
        getOutlet("androidPrompt").update({
          visible: true,
          message: "Completed",
          isDone: true,
        });
      }
    } else {
      // send password to NFC tags, so we can perform write operations
      handleException(
        "Tag not supported or tag already has password protection",
      );
    }
  } catch (error) {
    console.log("error:::::", JSON.stringify(error));
    handleException("Write Error");
  } finally {
    NfcManager.cancelTechnologyRequest();
    if (Platform.OS === "android") {
      setTimeout(() => {
        getOutlet("androidPrompt").update({
          visible: false,
          isDone: true,
        });
      }, 3000);
    }
  }
}