revtel/react-native-nfc-manager #93: Native module cannot be null

Rexima, 2018-10-06:
I got this error:
[image]

I just installed it with this command: **yarn add react-native-nfc-manager**
And put this line on my js file: **import NfcManager, {Ndef, NdefParser} from 'react-native-nfc-manager'**

whitedogg13, 2018-10-06:
Hi, when using `expo` (or `create-react-native-app`) with this library (or any library with native dependency), you will have to:

* first `eject`, to restore correct project structure with native support
* `react-native link react-native-nfc-manager` to link the native dependency

Thanks!

Rexima, 2018-10-07:
So i cannot use this with Expo?
I ejected it and than linked it, but it doesnt work, got the same error.

whitedogg13, 2018-10-09:
Hi, it should work with Expo. Please check #38 , which is also using Expo and reported to work.

I am personally not using Expo, but I will try that with this lib when I'm available. 

Thanks!

closed 2019-01-03 by whitedogg13