revtel/react-native-nfc-manager #95: [Document] write a section about how to use this library with Expo

whitedogg13, 2018-10-06:
Just a reminder, since many RN developers use `CRNA` or `Expo`, we should write a section to help them adapt this library.

christanfoden, 2018-10-18:
Would love this, or just some pointers about how it could be achieved at a glance. 

We're developing an app that requires NFC and wish to combine this with expo but there isn't much info available. Would you have to export using expo kit for example as I don't think you can edit info.Plist in expo without ejecting?

C0deMunk33, 2018-12-21:
I'm currently stuck with Expo, any hints?

[image]

whitedogg13, 2018-12-22:
@C0deMunk33 it seems that your expo project isn't successfully ejected, so the native module (`NativeNfcManager`) is still undefined. Maybe you can try this guide: 
* https://docs.expo.io/versions/v31.0.0/expokit/eject.

whitedogg13, 2019-01-03:
@C0deMunk33 If you have any updates, please share with us!

whitedogg13, 2019-01-04:
@C0deMunk33 Thanks for the solid information!

whitedogg13, 2019-10-24:
close inactive issue

closed 2019-10-24 by whitedogg13

troutgirl, 2019-11-07:
Do you still need a doc for this?

whitedogg13, 2019-11-07:
Yes, I believe it will still be helpful. It you can provide a PR, I will be happy to accept that. Thanks!

troutgirl, 2019-11-08:
Seems like it wouldn't be THAT hard to build a unimodule, which I think is the "right way" in Expo-think. I spoke to Brent from Expo team today and he said something like this is on their list but way down in priority. Thoughts?

whitedogg13, 2019-11-10:
Hi @troelsh thanks for contacting with expo team. I think currently we still need to guide expo user to properly `eject` to use 3rd party native module like this. 

About expo's unimodule, since most connectivity features such as NFC or BLE are hardware related, and not required by most apps, it's a natural approach to exclude them. Maybe they can do something like `expo-connectivity-modules` to include these native connectivity features, which might benefit app developers targeting IOT or custom hardware related stuff.

troelsh, 2019-11-10:
@whitedogg13  you mean @troutgirl right? :)

troutgirl, 2019-11-10:
Ah, good point! I will make sure to emphasize ejecting properly and what
you can/can't do in managed vs Bare workflow.

troutgirl, 2020-01-25:
Sorry for the delay, I had to switch from one library to another and it set
me back. Still working tho!

fr3fou, 2020-01-25:
what library did you switch to?

troutgirl, 2020-01-25:
Chirp.io!

niravbhimani53, 2020-09-02:
Hi, any update on how to use `react-native-nfc-manager` libary with [Expo](https://expo.io/)?