revtel/react-native-nfc-manager #96: Can't call setState on an unmounted component

emilany, 2018-10-09:
I followed the sample and added `this._stateChangedSubscription.remove();` on componentWillUnmount but I still am getting this error. Also, once this error appears I'm not able to scan an nfc again and have to run `react-native run-android` to make it work.

Do you possibly know of any workaround or am I doing something wrong? My entire code is below.

```
class Register extends Component {
  constructor(props) {
    super(props);
    this.state = {
      supported: true,
      enabled: false,
      tag: {},
    };
  }

  componentDidMount() {
    NfcManager.isSupported()
      .then((supported) => {
        this.setState({ supported });
        if (supported) {
          this.startNfc();
          this.startDetection();
        }
      });
  }

  componentWillReceiveProps(nextProps) {
    console.log('componentWillReceiveProps', nextProps);
    const { alertWithType } = this.props;
    if (nextProps.error) {
      alertWithType('error', 'Error', nextProps.error);
    }
    if (nextProps.status === 201) {
      alertWithType('success', 'Success', 'Successfully registered nfc');
      this.handleNavigation();
    }
  }

  componentWillUnmount() {
    if (this._stateChangedSubscription) {
      this._stateChangedSubscription.remove();
    }
  }

  handleNavigation = () => {
    const { navigation } = this.props;
    navigation.navigate('Authenticated');
  }

  startDetection = () => {
    NfcManager.registerTagEvent(this.onTagDiscovered)
      .then((result) => {
        console.log('registerTagEvent OK', result);
      })
      .catch((error) => {
        console.warn('registerTagEvent fail', error);
      });
  }

  onTagDiscovered = (tag) => {
    this.setState({ tag });
  }

  goToNfcSetting = () => {
    if (Platform.OS === 'android') {
      NfcManager.goToNfcSetting()
        .then((result) => {
          console.log('goToNfcSetting OK', result);
        })
        .catch((error) => {
          console.warn('goToNfcSetting fail', error);
        });
    }
  }

  onPressRegister = () => {
    const { tag } = this.state;
    const { dispatch } = this.props;
    dispatch(registerNfc(tag.id));
  }

  onPressEnable = () => {
    if (Platform.OS === 'android') {
      NfcManager.goToNfcSetting()
        .then((result) => {
          console.log('goToNfcSetting OK', result);
        })
        .catch((error) => {
          console.warn('goToNfcSetting fail', error);
        });
    }
  }

  startNfc() {
    NfcManager.start({
      onSessionClosedIOS: () => {
        console.log('ios session closed');
      },
    })
      .then((result) => {
        console.log('start OK', result);
      })
      .catch((error) => {
        console.warn('start fail', error);
        this.setState({ supported: false });
      });

    if (Platform.OS === 'android') {
      NfcManager.getLaunchTagEvent()
        .then((tag) => {
          console.log('launch tag', tag);
          if (tag) {
            this.setState({ tag });
          }
        })
        .catch((err) => {
          console.log(err);
        });

      NfcManager.isEnabled()
        .then((enabled) => {
          this.setState({ enabled });
        })
        .catch((err) => {
          console.log(err);
        });
    }
  }

  render() {
    const { tag, enabled } = this.state;
    return (
      <Nfc
        heading="Register NFC"
        description="Tap NFC to register"
        buttonText="Register NFC"
        onPressButton={this.onPressRegister}
        tag={tag.id}
        enabled={enabled}
        onPressEnable={this.onPressEnable}
      />
    );
  }
}

Register.propTypes = {
  navigation: PropTypes.object,
  dispatch: PropTypes.func,
  status: PropTypes.number,
  error: PropTypes.string,
  alertWithType: PropTypes.func,
};

const mapStateToProps = state => ({
  status: state.register.status,
  error: state.register.error,
});

export default connect(mapStateToProps)(connectAlert(Register));
```

whitedogg13, 2018-10-09:
Hi, sorry for the late response. In your code, I don't see the initial assignment for `this._stateChangedSubscription`, where does it come from?

emilany, 2018-10-10:
Thanks for responding. Managed to fix the problem by calling stopDetection during componentWillUnmount.

closed 2018-10-10 by emilany