revtel/react-native-nfc-manager #98: Write on MiFare Classic

chelinho139, 2018-10-10:
Hey, Ive been searching on code if there is support for writing data on Mifare Classic at the moment.
I see mifareClassicAuthenticate and mifareClassicReadBlock. Is there support for Mifare Write Block at the moment? 
Thanks

poison, 2018-10-10:
It should be fairly straightforward to add this.

@whitedogg13 a question for you: I just realize the following: the readBlock actually reads a whole sector, which is typically 4 blocks on a Mifare Classic 1K card. With writeBlock we don't want to enforce to write the whole sector of course, so we have the following options:

1. rename readBlock to readSector, and implement readBlock which reads only one block in a sector + implement utility methods like (blockToSector and sectorToBlock)
2. implement writeBlock that it writes only 1 block, and leave readBlock as-is

WDYT? My preference goes to the 1st, but it means that we're changing the API, which could be annoying for anybody who already implemented it. However, I think it's a necessary thing for consistency. Do you agree? I'll create a PR accordingly

whitedogg13, 2018-10-11:
@poison I also prefer the 1st option, let make it better!

poison, 2018-10-14:
Hey @chelinho139 , the PR is merged into master, please let us know if you experience any issues.

closed 2018-10-14 by poison