revtel/react-native-nfc-manager #102: Enhance isSupported to check for specific Technologies too

poison, 2018-10-18:
I've noticed on some Android devices MifareClassic is not enabled, which results in errors upon scanning a card. But it would be nice to know on forehand to see if a specific technology is available or not. Now the isSupported doesn't accept any parameters so my suggestion is to extend it so you can pass a TechRequest to it (like `isSupported(TechRequest.MifareClassic)`). Wouldn't change the default behaviour (if you pass nothing it will check if NFC in general is supported).

I can work on a PR, still need to do some testing first on other devices, but if you agree on this I'd make it blocking for releasing 1.0.

poison, 2018-10-18:
@whitedogg13 would be great to have your comments on this

whitedogg13, 2018-10-19:
@poison , There should be a property called `techTypes` in the detected in tag object (see line 59@Util.java ), so it should be possible to do things like this:

```javascript
NfcManager.registerTagEvent(tag => {
    if (tag.techTypes.find(tech => tech === NfcTech.XXX)) {
        // request tech handle
    } else {
        // do something else
    }
})
```

Maybe we should provide an enhanced document & example to address this?

poison, 2018-10-22:
Well I'm kinda busy now, I hope to get back to you by friday on this, but I already noticed on some devices that MifareClassic isn't supported on specific NFC chipsets (mainly the Broadcoms).

I found this piece of code that handles it nicely: [https://github.com/ikarus23/MifareClassicTool](https://github.com/ikarus23/MifareClassicTool/blob/3eb37b0afa3f57f0d18f2fcdfcee2435b47886d8/Mifare%20Classic%20Tool/app/src/main/java/de/syss/MifareClassicTool/Common.java#L588)

Sometimes you know it on forehand when it's not supported (like if your device has the broadcom chip), and sometimes you can only know it when you're scanning a card. My idea is to throw a specific error there so it's easy distinguishable from radio errors.

poison, 2018-11-17:
I created PR #117 , please let me know what you think of it. In my usecase I wanted to know the support at launch of the app, so before detecting a tag. The PR works for my usecase.

Best,
Nicolas

closed 2018-11-17 by poison