revtel/react-native-nfc-manager #103: mifareClassicWriteBlock fail issue

liew789, 2018-10-23:
In NfcManager.java, i saw a condition

>  else if (blockIndex >= mifareTag.getSectorCount()) {

, means that if my nfc tag have 16 sector , i cannot write data into block 16 or more. Is it a bug? blockIndex supposed have to compare with BlockCount (instead of SectorCount)right?

poison, 2018-10-23:
You're completely right! It's a cut&past error from a previous piece of code. I pushed a fix. Please let me know if it's resolved for you.

whitedogg13, 2018-10-23:
Hi @liew789 / @poison , thanks for issue reporting and fixing, a bug fix version 1.0.1 has just published!

poison, 2018-11-04:
Closing the issue since I tested the last fix and conform that you can write to blockIndex > sectorCount (wrote to block 60 on a 16-sector card)

closed 2018-11-04 by poison